Koenkk/zigbee2mqtt #16451: [New device support]:

pranavptl, 2023-02-01:
### Link

https://www.btf-lighting.com/collections/all-products/products/zigbee-3-0-wifi-2-4ghz-led-controller-dim-cct-rgb-rgbw-rgbcct-hue-bridge-tuya-dual-mode-gateway-smart-things?variant=43664435020002

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c1385a45a79243","nwkAddr":56536,"manufId":4417,"manufName":"_TZ3210_xwqng7ol","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1675255760786,"defaultSendRequestWhen":"immediate"}

### Comments

it pairs as router TS0502B - Miboxer FUT035Z

the trouble is that the power on behaviour doesn't work and does update status in Home Assistant after a power cycle

### Supported color modes

cct

patrickschoenherr, 2023-02-17:
@Koenkk 

Have the same problem as @pranavptl.
Could you please check if you can add support for Zigbee model 'TS0502B' and manufacturer name '_TZ3210_xwqng7ol' ?
Thank you.

patrickschoenherr, 2023-02-17:
@Koenkk 
Got the device working after updating to version 1.30.
Sorry for the previous comment.

closed 2023-03-28 by github-actions[bot]