Koenkk/zigbee2mqtt #16458: 1.30.0 fails to start

swests, 2023-02-01:
### What happened?

Ran updated script to update from 1.29.2 to 1.30.0
After update, Z2M fails to start

### What did you expect to happen?

Version to launch

### How to reproduce it (minimal and precise)

Update and launch

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

 0x26680700

### Adapter

ConBee2

### Debug log

> zigbee2mqtt@1.30.0 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/ignore-enoent.js:15
        if (er?.code !== 'ENOENT') {
               ^

SyntaxError: Unexpected token '.'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/rimraf-posix.js:14:28)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.30.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.30.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/simon/.npm/_logs/2023-02-01T16_35_36_312Z-debug.log

Koenkk, 2023-02-01:
What node js version are you using? (`node --version`), this should be at least 14.

swests, 2023-02-01:
Ah, that would be it then... v12.22.12. Will update and try again.

swests, 2023-02-01:
Is there a maximum node version?  E.g. can I go from 12 to 19 and be ok with Z2M

swests, 2023-02-02:
Node version issue. Resolved.

closed 2023-02-02 by swests