Koenkk/zigbee2mqtt #16460: MOES Thermostat program sensor - State max length is 255 characters

PeterCicaMolnar, 2023-02-01:
### What happened?

Moes BHT series Thermostat (BHT-002-GCLZB)
Program sensor is unavailable because of the error message in HA logger. Please see below.

### What did you expect to happen?

No errors like before in HA logger and available program sensor of the thermostat.

### How to reproduce it (minimal and precise)

Updated to 1.30.0

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

0x26680700

### Adapter

ConBee2

### Debug log

Logger: homeassistant
Source: core.py:1140
First occurred: 20:05:51 (66 occurrences)
Last logged: 20:17:51

Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fb952085770>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fb9520cd2a0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fb953bf1a10>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fb953c9e180>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fb953bf3140>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 696, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 556, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 673, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1480, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1140, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.moes_termosztat_program. State max length is 255 characters.

samourai64, 2023-02-02:
Hello
I have the same problem  with the same Thermostat
My config:
 Home Assistant 2023.2.0 
Supervisor 2023.01.1 
Operating System 9.5 
Interface utilisateur : 20230201.0 - latest 
Zigbee2MQTT Current version: 1.30.0-1
Adaptateur is SONOFF Zigbee 3.0 USB Dongle Plus

When i send an order to the thermostat i have this problem
`Logger: homeassistant
Source: core.py:1144
First occurred: 10:26:57 (98 occurrences)
Last logged: 11:21:17

Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f7b07ec00>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f7b07c040>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f7b07f7d0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f7b07c9e0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f7b07dee0>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 694, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 683, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1512, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1144, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.thermostat_program. State max length is 255 characters.
`

Thank's for the help

Mosher23, 2023-02-02:
Hello,
I confirm the problem. After update to 1.30 i experience same problem.
`Logger: homeassistant
Source: core.py:1144
First occurred: 11:54:20 (631 occurrences)
Last logged: 11:57:22

Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fe4e3565d90>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fe4df4c2ea0>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fe4df407760>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fe4e37e7a00>)
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...x7fe4e408fdf0>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 694, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 683, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1512, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1144, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.office_thermostat_program. State max length is 255 characters.
`

samourai64, 2023-02-03:
Hello
I observe 2 other porblem for me
- that now it's not possible for my thermostat to modify the program wiyh the zigbee2mtt web interface/Expose.
Il i made a modification it's not possible to save the change.
- My thermostat is always in preset mode "Hold". But now without modification, at each time of the weekdays program hours (i don't use it), it change the temperature cible with witch is in the programation.
I have 19° in the hold preset, and 4 times per days it change to the program t° 7°

Mysterious... If some on has solution.... Thank's

samourai64, 2023-02-06:
no bug fix for the moment....
Another users have this problem?
Thank's

Minims, 2023-02-06:
> raise InvalidStateError(
> homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.office_thermostat_program. State max length is 255 characters.

Yes I have this error too

Minims, 2023-02-06:
> * that's now  not possible for my thermostat to modify the program wiyh the zigbee2mtt web interface/Expose.
>   Il i made a modification it's not possible to save the change.

I confirmed the issue, too. unable to apply, my old program is visible in the state

osre77, 2023-02-07:
I also have an similar issue since 1.30.0
Every 80ms I get this error until after about 10 hours the system stops responding
`2023-02-06 21:41:32.000 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...xffff800673e0>)
Traceback (most recent call last):
File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
self._context.run(self._callback, *self._args)
File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 693, in _mqtt_handle_message
self._mqtt_data.state_write_requests.process_write_state_requests()
File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
entity.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 683, in _async_write_ha_state
self.hass.states.async_set(
File "/usr/src/homeassistant/homeassistant/core.py", line 1512, in async_set
state = State(
File "/usr/src/homeassistant/homeassistant/core.py", line 1144, in __init__
raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.thermostat_kinderzimmer_jan_program. State max length is 255 characters.`

KameDomotics, 2023-02-10:
At the moment the only solution is to downgrade Zigbee2mqtt (if you have a backup)

osre77, 2023-02-10:
I did downgrade to 1.29.2.
This solved that HA now works again, but Zigbee2mqtt still has issues, see:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/459

samourai64, 2023-02-21:
For me i prefer wait an update. i m not on the same place than my HA, so it's very diffcult to donwngard without risk....
Please, an update to correct this issue....
Thank's

csszokolai, 2023-02-26:
Not possible to modify the program on the zigbee2mqtt web interface/Expose tab. The Apply button is inactive.

samourai64, 2023-03-01:
Hello
Not fixed with the les version 1.30.2.1.
It's a big problem ...
I hope a solution

M3te0r, 2023-03-07:
Hi, I'm facing this problem too
Thermostat : Moes BHT series Thermostat (BHT-002-GCLZB)
Home Assistant : 2023.3.1
Zigbee2MQTT : 1.30.2-1

Here's the traceback from log viewer

```
2023-03-07 12:24:36.148 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl... 0x7f8c3589e0>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 722, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 559, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 676, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1537, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1186, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.thermostat_program. State max length is 255 characters.
```

j-f-d, 2023-03-10:
1. My take on this is that components/mqtt/models.py shouldn't call entity.async_write_ha_state() with a state longer than MAX_LENGTH_STATE_STATE, because it can't prevent an MQTT client from posting a longer message.
2. In helpers/entity.py, async_write_ha_state() should cope gracefully if it is called with a state longer than MAX_LENGTH_STATE_STATE.
3. If Z2M is to work well with HA and not cause strange problems, it should probably also limit MQTT messages to MAX_LENGTH_STATE_STATE. I.e. another solution is required for handling program messages.

j-f-d, 2023-03-10:
> At the moment the only solution is to downgrade Zigbee2mqtt (if you have a backup)

Switching to ZHA works, so may also be a work-around for some.

Koenkk, 2023-03-12:
Can someone provide the debug log when the state of this device is published?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

samourai64, 2023-03-12:
Hello
If someone arrived to have the debbug logging. i don't arrived
I don't know where to insert 
advanced:
  log_level: debug

osre77, 2023-03-12:
I might remember wrong, but I think I tried this and did not get a single extra line of logging.

M3te0r, 2023-03-12:
Hi @Koenkk, I enabled the debug log, I think this is what you're looking for ?

Zigbee2MQTT:info  2023-03-12 18:03:08: MQTT publish: topic 'zigbee2mqtt/Thermostat', payload '{"child_lock":"UNLOCK","current_heating_setpoint":20,"deadzone_temperature":null,"heat":"OFF","linkquality":255,"local_temperature":21.5,"local_temperature_calibration":-4,"max_temperature":45,"max_temperature_limit":null,"min_temperature_limit":1,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'

danielszilagyi, 2023-03-12:
> Can someone provide the debug log when the state of this device is published?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Hi @Koenkk, I removed one from the network and joined it with debug enabled, hope it helps revealing the issue.
I have had a filter previously configured for this device not to publish "program" to mqtt though, to stop it from choking. 
So if a new debug log is required with this filter removed, let me know.

```
Zigbee2MQTT:info  2023-03-13 00:43:24: Device 'Eloszoba_Floor' joined
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Eloszoba_Floor","ieee_address":"0x50325ffffe289f0c"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Eloszoba_Floor"},"type":"device_connected"}'
Zigbee2MQTT:info  2023-03-13 00:43:25: Starting interview of 'Eloszoba_Floor'
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Eloszoba_Floor","ieee_address":"0x50325ffffe289f0c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Eloszoba_Floor"},"type":"pairing"}'
Zigbee2MQTT:debug 2023-03-13 00:43:25: Device 'Eloszoba_Floor' announced itself
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Eloszoba_Floor","ieee_address":"0x50325ffffe289f0c"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-03-13 00:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Eloszoba_Floor"},"type":"device_announced"}'
Zigbee2MQTT:debug 2023-03-13 00:43:25: Received Zigbee message from 'Eloszoba_Floor', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_aoclfnxz","modelId":"TS0601"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-03-13 00:43:25: Received Zigbee message from 'Eloszoba_Floor', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-03-13 00:43:26: Received Zigbee message from 'Eloszoba_Floor', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-03-13 00:43:26: Received Zigbee message from 'Eloszoba_Floor', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-03-13 00:43:26: Successfully interviewed 'Eloszoba_Floor', device has successfully been paired
Zigbee2MQTT:info  2023-03-13 00:43:26: Device 'Eloszoba_Floor' is supported, identified as: Moes Moes BHT series Thermostat (BHT-002-GCLZB)
Zigbee2MQTT:info  2023-03-13 00:43:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Moes BHT series Thermostat","exposes":[{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":3,"description":"The delta between local_temperature and current_heating_setpoint to trigger Heat","name":"deadzone_temperature","property":"deadzone_temperature","type":"numeric","unit":"°C","value_max":5,"value_min":0,"value_step":1},{"access":3,"description":"Maximum temperature limit. Cuts the thermostat out regardless of air temperature if the external floor sensor exceeds this temperature. Only used by the thermostat when in AL sensor mode.","name":"max_temperature_limit","property":"max_temperature_limit","type":"numeric","unit":"°C","value_max":35,"value_min":0},{"access":3,"description":"Minimum temperature limit for frost protection. Turns the thermostat on regardless of setpoint if the tempreature drops below this.","name":"min_temperature_limit","property":"min_temperature_limit","type":"numeric","unit":"°C","value_max":5,"value_min":1},{"features":[{"access":3,"description":"Temperature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":35,"value_min":5,"value_step":1},{"access":1,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":3,"description":"Offset to be used in the local_temperature","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C","value_max":30,"value_min":-30,"value_step":0.1},{"access":3,"description":"Mode of this device","name":"system_mode","property":"system_mode","type":"enum","values":["off","heat"]},{"access":1,"description":"The current running state","name":"running_state","property":"running_state","type":"enum","values":["idle","heat","cool"]},{"access":3,"description":"Mode of this device (similar to system_mode)","name":"preset","property":"preset","type":"enum","values":["hold","program"]}],"type":"climate"},{"access":3,"description":"Select temperature sensor to use","name":"sensor","property":"sensor","type":"enum","values":["IN","AL","OU"]},{"access":3,"description":"Time of day and setpoint to use when in program mode","features":[{"access":3,"name":"weekdays_p1_hour","property":"weekdays_p1_hour","type":"nume [... cut ...] rsion":1,"ieeeAddr":"0x50325ffffe289f0c","manufacturerID":4098,"manufacturerName":"_TZE200_aoclfnxz","model":"BHT-002-GCLZB","networkAddress":53406,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"heat":"OFF","linkquality":117,"local_temperature":22,"local_temperature_calibration":-3,"max_temperature":35,"max_temperature_limit":32,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","running_state":"idle","sensor":"AL","system_mode":"heat"}'
```

Koenkk, 2023-03-13:
The state is long because of the `program`.

As a workaround you can filter the `program` attribute using `filtered_attributes` (https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options) but ideally HA should allow for longer states.

osre77, 2023-03-13:
if 255 is really the maximum for the MQTT message content, then tis is way too short. Even without the `program` part I think the message is longer than 255 characters.
Does anyone know if this 255 character limit was introduced with HA 2023.2 ?

j-f-d, 2023-03-14:
> if 255 is really the maximum for the MQTT message content, then tis is way too short. Even without the `program` part I think the message is longer than 255 characters. Does anyone know if this 255 character limit was introduced with HA 2023.2 ?

It might be around May 25th 2021, This [commit](https://github.com/home-assistant/core/commit/58e37435b3396ca2e3a084597bf5335772fb2b30) is when the valid_state() function in core.py was changed from using a length of 256 to MAX_LENGTH_STATE_STATE.

PeterCicaMolnar, 2023-03-14:
Hi @Koenkk!
First of all, thanbk you for the Z2M, this great piece of software!

Regarding this thermostat I would say, that's really great you introduced the option to program the thermostat from Z2M GUI.
From the other hand, this option is not used daily. I can say many of us just set up some programs on the thermostat and done.
What I ususally do is set to hold if we are not at home for longer time or just set back to program mode to let it run. I did not touched the programs in the last 2 years. I would be surprised if someone change the programs frequently.
I would suggest here a switch or an option to supress the programming option on the GUI or somewhere if it's not a big deal to implement it.
Just an idea :)
Thank you again and have a great day,
Peter

samourai64, 2023-04-01:
Hello
Problem not solved in the last update 1.30.3....

KameDomotics, 2023-04-02:
I'm still using Zigbee2mqtt version 1.29.2-1, it's the latest version that doesn't have this problem.

KameDomotics, 2023-05-01:
I don't think it's ONLY a limitation of Home Assistant, with the latest version of HA (2023.4.6) and an old version of Z2M (1.29.2-1) this problem doesn't exist. Z2M 1.29.2-1 is the latest version that doesn't have this problem.

osre77, 2023-05-01:
I switched to ZHA now. It does not support all features, like child-lock, but at least it works.

danielszilagyi, 2023-05-01:
> I switched to ZHA now. It does not support all features, like child-lock, but at least it works.

There was a very simple workaround mentioned on this thread by Koenkk earlier, the program schedule setting of course won’t be usable, but the thermostat will work after applying it.

https://github.com/Koenkk/zigbee2mqtt/issues/16460#issuecomment-1466518618

samourai64, 2023-05-01:
hello
Ca you explain how t o apply this  workaround please
In Parameters i have 
- Attributs publish filtrés
- Attributs filtrés du cache
- Attributs optimistes filtrés
And how write the filter: "Program" [Program] ....
My HA is not in my house and i dont want "brick" my instalation

Thank's for your help

miklosbagi, 2023-05-03:
I'm not sure on the FR version unfortunately, in english, it looks like:

1. Go to dashboard and find your termostat device, click it
1. Go to settings, and find `Filtered publish attributes`, click `+`
1. Type `program` like that
1. Hit `Submit` on the bottom of the page to save this.

That is it.

> but ideally HA should allow for longer states.
Word.  
I've got way too many `{{ ns.names| truncate (254,true) }}`s in my config.

j-f-d, 2023-05-03:
In the Z2M configuration.yaml, I have:
```
    friendly_name: Palm Thermostat
    debounce: 1
    filtered_attributes:
      - program
    homeassistant: {}
    legacy: false
    optimistic: true

```

In HA, I also had to disable the 'Program' sensor in the MQTT Device page to prevent an annoying message in the HA System Log.

KameDomotics, 2023-05-07:
> 

`  
  friendly_name: Termostato
  debounce: 1
  optimistic: true
  filtered_attributes:
    - program
  homeassistant: {}
  filtered_cache:
    - program `
    
    
It's correct?

Koenkk, 2023-05-07:
@miklosbagi the `truncate` is a very elegant solution, I've added a default `truncate(254, True, '', 0)` for the `program` sensor now. So no need to apply a manual fix anymore

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gorbunovav, 2023-06-11:
@Koenkk hmm. It seems that on HA 2023.6.1 if  the program value is `null` this template fails with:

```
File "/usr/local/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate
  if len(s) <= length + leeway:
    ^^^^^^
 
TypeError: object of type 'NoneType' has no len()
```

thargy, 2023-07-08:
I still get this exact same error in 1.32.1
```
08T20:04:17.080Z","linkquality":98,"local_temperature":24.5,"local_temperature_calibration":-1,"max_temperature_limit":30,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":"IN","system_mode":"heat"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 292, in message_received
    _update_state(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 243, in _update_state
    payload = self._template(msg.payload, PayloadSentinel.DEFAULT)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value
    rendered_payload = self._value_template.async_render_with_possible_json_value(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 735, in async_render_with_possible_json_value
    return _render_with_context(self.template, compiled, **variables).strip()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2179, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "
  File "/usr/local/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate
    if len(s) <= length + leeway:
       ^^^^^^
TypeError: object of type 'NoneType' has no len()
```

Happens every 7 seconds.

csszokolai, 2023-07-16:
I tried to "reprograming" the device directly with it's buttons. I just tap through until the end the earlier saved program. After that I checked what I see in the z2m and it recovered! Now I see the whole program.

closed 2023-08-26 by github-actions[bot]

SSG1111, 2023-09-04:
issue is still valid

Spica2000, 2024-09-30:
I also have same issue with MOES Thermostat [BHT-002/BHT-006]
HA 2024.9.3
Zigbee2MQTT Current version: 1.40.1-1
Mosquitto broker Current version: 6.4.1

The only temporal workaround like posted here: [(https://github.com/Koenkk/zigbee2mqtt/issues/19788#issuecomment-2369919048)]

Log:

2024-09-28 15:24:22.646 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Cannot update state for entity sensor.floorheatredbathroom_program after processing payload on topic zigbee2mqtt/FloorHeatRedBathroom. The requested state (941) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'saturday_p1_hour': 6, 'saturday_p1_minute': 0, 'saturday_p1_temperature': 24, 'saturday_p2_hour': 12, 'saturday_p2_minute': 0, 'saturday_p2_temperature': 23, 'saturday_p3_hour': 14, 'saturday_p3_minute': 30, 'saturday_p3_temperature': 22, 'saturday_p4_hour': 17, 'saturday_p4_minute': 30, 'saturday_p4_temperature': 21, 'sunday_p1_hour': 6, 'sunday_p1_minute': 0, 'sunday_p1_temperature': 19, 'sunday_p2_hour': 12, 'sunday_p2_minute': 30, 'sunday_p2_temperature': 20, 'sunday_p3_hour': 14, 'sunday_p3_minute': 30, 'sunday_p3_temperature': 21, 'sunday_p4_hour': 18, 'sunday_p4_minute': 30, 'sunday_p4_temperature': 20, 'weekdays_p1_hour': 6, 'weekdays_p1_minute': 0, 'weekdays_p1_temperature': 20, 'weekdays_p2_hour': 11, 'weekdays_p2_minute': 30, 'weekdays_p2_temperature': 21, 'weekdays_p3_hour': 13, 'weekdays_p3_minute': 30, 'weekdays_p3_temperature': 22, 'weekdays_p4_hour': 17, 'weekdays_p4_minute': 30, 'weekdays_p4_temperature': 23}
2024-09-28 15:24:22.647 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '{'saturday_p1_hour': 6, 'saturday_p1_minute': 0, 'saturday_p1_temperature': 24, 'saturday_p2_hour': 12, 'saturday_p2_minute': 0, 'saturday_p2_temperature': 23, 'saturday_p3_hour': 14, 'saturday_p3_minute': 30, 'saturday_p3_temperature': 22, 'saturday_p4_hour': 17, 'saturday_p4_minute': 30, 'saturday_p4_temperature': 21, 'sunday_p1_hour': 6, 'sunday_p1_minute': 0, 'sunday_p1_temperature': 19, 'sunday_p2_hour': 12, 'sunday_p2_minute': 30, 'sunday_p2_temperature': 20, 'sunday_p3_hour': 14, 'sunday_p3_minute': 30, 'sunday_p3_temperature': 21, 'sunday_p4_hour': 18, 'sunday_p4_minute': 30, 'sunday_p4_temperature': 20, 'weekdays_p1_hour': 6, 'weekdays_p1_minute': 0, 'weekdays_p1_temperature': 20, 'weekdays_p2_hour': 11, 'weekdays_p2_minute': 30, 'weekdays_p2_temperature': 21, 'weekdays_p3_hour': 13, 'weekdays_p3_minute': 30, 'weekdays_p3_temperature': 22, 'weekdays_p4_hour': 17, 'weekdays_p4_minute': 30, 'weekdays_p4_temperature': 23}' from 'zigbee2mqtt/FloorHeatRedBathroom'

maciekdnd, 2024-09-30:
@Spica2000 if you don' t need "program" you may exclude it from the mqtt message. Open device settings in Z2M, go to settings and add "program" to Filter attributes section. Should help. If you need that part, you have to probably wait.

Spica2000, 2024-10-01:
> @Spica2000 if you don' t need "program" you may exclude it from the mqtt message. Open device settings in Z2M, go to settings and add "program" to Filter attributes section as others already mentioned it. Should help. If you need that part, you have to probably wait.

Thank you for your comment. I don't understand completely what this filter actually does. Seems something with program mode of the thermostat.

maciekdnd, 2024-10-01:
@Spica2000 On actual device you have program memory to keep all your settings for the week. This is a lot of data to transfer in one message on the mqtt protocol. If you don't need to see or change those specific settings on the HA dashboard, you can filter whole "program" from being transferred in every message from your thermostat to Z2M and HA. If you filter it (which means you don't have access on Z2M to those values, and in most cases you don't need to) then you don't overflow your message with big payload and you don't get this warning.

samourai64, 2024-10-01:
Hello
I have the same probleme but my z2mqtt is in french. So can you confirm witch is the flter to apply with program. First? Second or the two?
[image]

M3te0r, 2024-10-01:
Bonjour (Hello)!
The setting in french is under "Attributs publish filtrés", a bit upper
[image]