Koenkk/zigbee2mqtt #16463: 1.30 broke ID Lock 150 integration (Datek)

epkulse, 2023-02-01:
### What happened?

Upgraded from 1.29.2 to 1.30.0. After upgrade, ID Lock 150 state changed to "unsupported". Had to revert to 1.29 again.

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20210708

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

owangen, 2023-02-01:
PR with fix https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425

Koenkk, 2023-02-02:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-02 by Koenkk