Koenkk/zigbee2mqtt #16464: [New device support]: Benexmart bulb

dbertie1, 2023-02-01:
### Link

https://www.aliexpress.com/item/1005002347074353.html?spm=a2g0o.order_detail.order_detail_item.3.3be2f19cmKN8Kl

### Database entry

{"dateCode":"","description":"-","friendly_name":"0xb4e3f9fffea762c4","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffea762c4","lastSeen":1675104515940,"manufacturerID":4098,"manufacturerName":"_TZ3210_1qismabg","model":"TS0502B","modelID":"TS0502B","networkAddress":16297,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"}

### Comments

I bought a pack of three zigbee bulbs and whilst they paired, they weren't recognised - being model: TS0502B but with an unrecognised manufacturer name of '_TZ3210_1qismabg'.

After a bit of playing around I have found that they work exactly the same as the other TS0502B devices (albeit they are bulbs, not batten controller).

I managed to get it working by adding this to existing definitions I found in one of the files!

{
        zigbeeModel: ['TS0502B'],
        model: 'TS0502B',
        vendor: 'TuYa',
        description: 'Light controller',
        whiteLabel: [
            {vendor: 'Mercator Ikuü', model: 'SMI7040', description: 'Ford Batten Light'},
            {vendor: 'Mercator Ikuü', model: 'SMD9300', description: 'Donovan Panel Light'},
            {vendor: 'Mercator Ikuü', model: '_TZ3210_1qismabg', description: 'Bulb Light'},
        ],
        extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 500], disablePowerOnBehavior: true}),
    }

I hope this helps and apologies, this was a bit hamfisted, but I really struggled with the generic instructions to write an extension.

David

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =
{
        zigbeeModel: ['TS0502B'],
        model: 'TS0502B',
        vendor: 'TuYa',
        description: 'Light controller',
        whiteLabel: [
            {vendor: 'Mercator Ikuü', model: 'SMI7040', description: 'Ford Batten Light'},
            {vendor: 'Mercator Ikuü', model: 'SMD9300', description: 'Donovan Panel Light'},
            {vendor: 'Mercator Ikuü', model: '_TZ3210_1qismabg', description: 'Bulb Light'},
        ],
        extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 500], disablePowerOnBehavior: true}),
    }

module.exports = definition;
```

Koenkk, 2023-02-02:
With z2m 1.29.2 all `TS0502B` are supported (regardless of the `manufacturerName`)

closed 2023-02-02 by Koenkk

dbertie1, 2023-02-02:
Aha, and thanks, it seems I am a log way behind the times.

Now to find out why my docker based version isn’t up to date!

Thanks again,

David