Koenkk/zigbee2mqtt #16467: 1.30 breaks Tuya TS0601 _TZE200_lu01t0zl Human presence sensor with fall function

mescalum, 2023-02-01:
### What happened?

after upgrading the device presence sensor show as unknown in Homeassistant. After restore everyting works again.

### Zigbee2MQTT version

1.30

### Adapter firmware version

not sure 

### Adapter

Slaesh stick

brot123, 2023-02-02:
i have also Problems with 1.30 with my TS601 lu01t0zl, what exactly logs you need about it? The Sensor didnt work ,i could change to 1.30 tomorrow again, i must change before my wife come back, you know:)

alexebner, 2023-02-02:
Same Here with     TS0601_TZE200_ztc6ggyl

It logs something like report 0 but only 1,2 allowed.
But only in a short red popup. not ind the logs.
I also have rolled back because my wife is killing me when the light in the bathroom is not working.

cdemi, 2023-02-02:
I confirm I have this problem as well

mescalum, 2023-02-03:
@alexebner , I have the Problem with my wife in the dark bathroom too. What makes it worse that Z2M keeps updating itself without any prompts and then issue reoccurs. Happened twice since the first rollback. I have now deactivated the update.

cdemi, 2023-02-03:
I think Koenkk/zigbee-herdsman-converters/pull/4859 is the cause of this issue

cdemi, 2023-02-03:
Actually, it looks like problem is not with the hedsman converter, because I see the device reporting Presence and Occupation correctly in Zigbee2MQTT
[image]

However Home Assistant is treating these as Unknown
[image]

mescalum, 2023-02-03:
yes, thanks for illustrating. Same here. Sensors such as moving forward are shown correctly. It is just those two unknown sensors,which are unfortunately the important ones.

alexebner, 2023-02-04:
Today i have updatet all again and flashed the new firmware on my Sonoff stick but the presence detector does not work.
The Log says:
Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of 1,2}

alexebner, 2023-02-04:
But this topic reports the sensors.

zigbee2mqtt/PM Küche:
{"fall_down_status":"none","fall_sensitivity":6,"illuminance_lux":32,"linkquality":54,"motion_direction":"standing_still","motion_speed":1,"occupancy":"false",**"presence":"true"**,"radar_scene":"area","radar_sensitivity":8,"static_dwell_alarm":null,"tumble_alarm_time":2,"tumble_switch":1}

{"fall_down_status":"none","fall_sensitivity":6,"illuminance_lux":32,"linkquality":54,"motion_direction":"standing_still","motion_speed":0,"occupancy":"false","**presence":"false**","radar_scene":"area","radar_sensitivity":8,"static_dwell_alarm":null,"tumble_alarm_time":2,"tumble_switch":1}

And ther it is reporting as it should.

alexebner, 2023-02-04:
I have found another topic:
homeassistant/binary_sensor/pm_kuche_presence/presence 

Now i have changed the automation to MQTT topic Change and it looks like its working.

trigger:
  - platform: mqtt
    topic: homeassistant/binary_sensor/pm_kuche_presence/presence
    payload: '"true"'

Koenkk, 2023-02-04:
The problem is that true became a string, `"true"` instead of `true`, I've pushed a fix, let me know if this works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

alexebner, 2023-02-04:
I am not able to test it because the edge brunsh wont start at my homeasisstant.

When i insert my config and press save then the configuration page does get blank and i am unable to start Z2M.

When i switch back to normal brunsh it works again.

mescalum, 2023-02-05:
@Koenkk this fixes the issue for Presence which is the important one. Can you do the same trick for Occupancy which still shows as unknown in HA? Thank you very much.

Koenkk, 2023-02-05:
@mescalum it should already be fixed for `occupancy`, maybe it takes some time before the sensor updates its `occupancy`?

mescalum, 2023-02-05:
@Koenkk Z2MEdge shows occupancy as null, even after refreshing edge and repairing. I guess that is why HA shows it still as unknown.

Koenkk, 2023-02-05:
@mescalum has the occupancy ever worked before? Maybe it only triggers in very specific sitatuions.

xistor4, 2023-02-05:
> @mescalum has the occupancy ever worked before? Maybe it only triggers in very specific situations.

The occupancy sensor behaved like a normal motion sensor triggering instantly from motion, where the presence sensor was slow to react but more accurate on continued room presence

mescalum, 2023-02-05:
@Koenkk it shows value Occupied in z2m v1.29 and prior at least. I think it also shows that in v1.30 but am not going to verify that currently. Our bathroom window openend last time while wife was in shower as humidity was going up and no presence reported, she cried for help as ot is facing the road.

cdemi, 2023-02-05:
From my end occupancy definitely worked because I have automations setup on that sensor

robertbigk, 2023-02-05:
Is any one can say how to fix that now i all house light not work and i dont know how fix issue

robertbigk, 2023-02-05:
@alexebner Sorry i not much undestood what You mean second trigger?
[image]
[image]

Can You explane me what to do in long version?:)

alexebner, 2023-02-05:
[image]

alexebner, 2023-02-05:
To get all MQTT messages i am using the windows software MQTT.fx

robertbigk, 2023-02-05:
@alexebner 
[image]
[image]

Sorry I ask, but i miss something? I try that and not work

xistor4, 2023-02-06:
For anyone else having issues but also using node-red with home assistant,I was able to get it working fine with the zigbee2mqtt integration with node red

Koenkk, 2023-02-06:
Can anybody else confirm that this issue is fixed in the latest dev? Then I can create a hotfix release.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

alexebner, 2023-02-06:
@robertbigk 

You have to search for the MQTT Topic with the name instead of the ID because the Problem is that the ID/presence ist not availabe.

robertbigk, 2023-02-06:
@Koenkk Hi , I got two sensors MIR-HE200-TY. Install Zigbee2MQTT Edge in home assistant and confirm all devices back to life and report state in home assistant.

closed 2023-02-06 by Koenkk

mescalum, 2023-02-06:
@Koenkk, sorry, am late to respond. For me the presence sensor is fixed in EDGE 1.30.1-dev and my automations work. Thanks for that. However, the occupancy sensor is not. It shows as Null in Z2M GUI and as unavailable in HA GUI. I am happy to create a new issue for that. Maybe someone can confirm the descibed behavior? For me it is no longer critical as I dont use occupancy sensor.

alexebner, 2023-02-06:
Same behaviour here
Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'from')}

And my workarround over the MQTT does not work anymore.

cdemi, 2023-02-06:
[image]

Yes, it looks like occupation is still Unknown

xistor4, 2023-02-07:
has broken the occupancy sensor for me also with " Exception while calling
fromZigbee converter: Cannot read properties of undefined (reading 'from')}"

raphaabreu, 2023-02-07:
I just updated in hoping of seeing this working again but the occupancy sensor still shows as unkown.

I can see that both values seem to have different types now in mqtt:
[image]

While the entity is configured to match the payload to boolean types instead of strings:

[image]

Koenkk, 2023-02-07:
@xistor4 could you provide the debug log of this?