Koenkk/zigbee2mqtt #16469: "Error while starting zigbee-herdsman" after restart

manuel-mauky, 2023-02-02:
### What happened?

zigbee2mqtt was working perfectly fine but after a restart of the computer it doesn't start anymore. I haven't changed anything in the config.
I'm using docker-compose and get this error message (log level is debug):

```
debug 2023-02-02 00:55:21: Loaded state from file /app/data/state.json
info  2023-02-02 00:55:21: Logging to console and directory: '/app/data/log/2023-02-02.00-55-21' filename: log.txt
debug 2023-02-02 00:55:21: Removing old log directory '/app/data/log/2023-02-02.00-52-46'
info  2023-02-02 00:55:21: Starting Zigbee2MQTT version 1.30.0 (commit #8f781db)
info  2023-02-02 00:55:21: Starting zigbee-herdsman (0.14.89)
debug 2023-02-02 00:55:21: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
error 2023-02-02 00:55:32: Error while starting zigbee-herdsman
error 2023-02-02 00:55:32: Failed to start zigbee
error 2023-02-02 00:55:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-02-02 00:55:32: Exiting...
error 2023-02-02 00:55:32: undefined
```

I've checked all instructions in the "failed-to-start" manual.

docker-compose.yml:
```
version: "3"
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ./zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
```

configuration.yaml:
```
permit_join: true
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.188.39
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
advanced:
  network_key:
    - 55
    - 75
    - 13
    - 146
    - 45
    - 72
    - 102
    - 115
    - 177
    - 50
    - 85
    - 61
    - 240
    - 244
    - 161
    - 126
  log_level: debug
```

### What did you expect to happen?

zigbee2mqtt should start normally without issues

### How to reproduce it (minimal and precise)

I'm not sure. I didn't change anything in the settings.

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

2.19.03

### Adapter

conbee 2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10563183/log.txt)

cloudbr34k84, 2023-02-02:
i dont have the same issue, but this is getting frustrating

Koenkk, 2023-02-02:
Try with:

```
serial:
  port: /dev/ttyACM0
  type: deconz
```

lerra, 2023-02-12:
I'm having the same issue with the latest docker container, everything was ok before the update

lerra, 2023-02-12:
After almost two days of a dark house I managed to figure it out by checking if there was another process locking / using the device, and yes, there was one. Run the following: `sudo lsof |grep -i <devicename>`
Just make sure to stop your zigbee2mqtt when you do this.

closed 2023-03-23 by github-actions[bot]