Koenkk/zigbee2mqtt #16470: Again after updating: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

cloudbr34k84, 2023-02-02:
### What happened?

After updating to the latest version, im getting Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

Background
I have 141 devices, large home, no idea if that make any difference, but the latest update was to support larger networks?

### Zigbee2MQTT version

 Zigbee2MQTT version 1.30.0 (commit #unknown)

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20221226.zip

### Adapter

Sonoff Coordinator P

cloudbr34k84, 2023-02-02:
@Koenkk thank you for everything you do with Z2<. Was wondering if you could help. I updated to the latest version and i updated the firmware to 20221226. Now Z2m is not starting. I have looked at the help guide link for this but nothing is helping. Im looking at the hardware and the port hasn't changed. Any help or guidance would be appreciated. I dont wwant to have to move back to ZHA, but its been so unstable for me. 

One issue i been seeing is Network Table Full?? Whats that about

Anyway not sure if you can help.

```
ttyUSB0
/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_ea398510f319ec11b7c776e5f01c6278-if00-port0
Subsystem:
tty
Device path:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_ea398510f319ec11b7c776e5f01c6278-if00-port0
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_ea398510f319ec11b7c776e5f01c6278-if00-port0
  /dev/serial/by-path/pci-0000:00:14.0-usb-0:3:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: Sonoff_Zigbee_3.0_USB_Dongle_Plus
ID_MODEL_ENC: Sonoff\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus
ID_MODEL_ID: ea60
ID_PATH: pci-0000:00:14.0-usb-0:3:1.0
ID_PATH_TAG: pci-0000_00_14_0-usb-0_3_1_0
ID_REVISION: '0100'
ID_SERIAL: ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_ea398510f319ec11b7c776e5f01c6278
ID_SERIAL_SHORT: ea398510f319ec11b7c776e5f01c6278
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_VENDOR: ITead
ID_VENDOR_ENC: ITead
ID_VENDOR_ID: 10c4
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '12289032959'
```

```
data_path: /config/zigbee2mqtt
socat:
  enabled: true
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
  zigbee_herdsman_debug: true
mqtt:
  base_topic: zigbee2mqtt
  user: xxxxxx-user
  password: xxxxxxxxx
  server: mqtt://core-mosquitto:1883
serial:
  port: /dev/ttyUSB0
availability: true
advanced:
  log_level: info
  pan_id: GENERATE
  ext_pan_id:
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
  network_key: GENERATE
  channel: 25
homeassistant: true
discovery_topic: homeassistant
status_topic: hass/status
```

dragondagda, 2023-02-02:
@cloudbr34k84 Have you tried flashing it again? Had the same issue, but after reflashing the firmware 20221226 zigbee2mqtt is starting

cloudbr34k84, 2023-02-02:
> @cloudbr34k84 Have you tried flashing it again? Had the same issue, but after reflashing the firmware 20221226 zigbee2mqtt is starting

I can try flashing again?? did you have the same error message??

cloudbr34k84, 2023-02-02:
> @cloudbr34k84 Have you tried flashing it again? Had the same issue, but after reflashing the firmware 20221226 zigbee2mqtt is starting

which firmware did you use??

[52P2_CC2652P_launchpad_coordinator_20221226.zip]

or

[CC1352P2_CC2652P_other_coordinator_20221226.zip]

cloudbr34k84, 2023-02-02:
> @cloudbr34k84 Have you tried flashing it again? Had the same issue, but after reflashing the firmware 20221226 zigbee2mqtt is starting

No go unfortunately! :(

dragondagda, 2023-02-02:
@cloudbr34k84 I have the Electrolama zzh! adapter and flashed CC2652R_coordinator_20221226.zip

I also had the log "Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)" after starting zigbee2mqtt 1.30.0 and my first attempt to flash the firmware on the adapter. After reading https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html I decided to retry flashing the 20221226 firmware. Afterwards it worked. If it hadn't I would have reflashed the old firmware 20220219

cloudbr34k84, 2023-02-02:
> @cloudbr34k84 I have the Electrolama zzh! adapter and flashed CC2652R_coordinator_20221226.zip
>
> I also had the log "Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)" [...]

Hey should i reflash the reflash source routing?
[CC2530_CC2591_SOURCE_ROUTING_20211116.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2530_CC2591_SOURCE_ROUTING_20211116.zip)

dragondagda, 2023-02-02:
@cloudbr34k84 No, don't do that. That is a firmware for another chip. If a reflash hasn't worked either double check the [flashing instructions for your adapter](https://www.zigbee2mqtt.io/guide/adapters/#recommended) (perhaps you did miss something) or for now flash the old 20220219 firmware.

cloudbr34k84, 2023-02-02:
> @cloudbr34k84 No, don't do that. That is a firmware for another chip. If a reflash hasn't worked either double check the [flashing instructions for your adapter](https://www.zigbee2mqtt.io/guide/adapters/#recommended) (perhaps you did miss something) or for now flash the old 20220219 firmware.

yep its flashing correctly using Texas Instruments. This is so frustrating

cloudbr34k84, 2023-02-02:
> yep its flashing correctly using Texas Instruments. This is so frustrating

i have flashed 20220219 firmware and still nothing. i should never have updated Z2m... im going to have to go to ZHA, just is just crazy.

dragondagda, 2023-02-02:
does the adapter work with the firmare 20220219 and zigbee2mqtt version 1.29.2 ?

cloudbr34k84, 2023-02-02:
> does the adapter work with the firmare 20220219 and zigbee2mqtt version 1.29.2 ?

How do you downgrade?? The instructions aren't clear?

dragondagda, 2023-02-02:
It depends how you installed zigbee2mqtt. With which installation method have you installed Zigbee2Mqtt? https://www.zigbee2mqtt.io/guide/installation/

cloudbr34k84, 2023-02-02:
I ended up going back to 2 days using my backup.

closed 2023-02-03 by cloudbr34k84