Koenkk/zigbee2mqtt #16471: _TZE200 known as TS0601_thermostat_3

MurksLP, 2023-02-02:
### What happened?

Is there a way to increase the local_temperature_calibration? The currently available offsets are -3 to +3, the older devices support from -10 to +10. Thats a big issue if the Device is mounted on a heater under a Window.

### What did you expect to happen?

A Value from -10 to +10

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Silvercrest Zigbee Hub, eszp

theonlyzby, 2023-02-03:
I am running version 1.29.1-dev commit: bf339636, and mine cannot go below -12.7°C. And personnaly I expect (and would like) to go way below this, because the valve is installed into a decorative radiator cover, and delta when heating from cold temperature can go as high as 16-17 °C!
The slider offers upt to -30°C, no??

DennisVonDerBey, 2023-02-22:
I am facing the same issue. Our heater is placed in a niche under the window and the delta is like 15 degrees celsius. Is there any technical limitation at play for the devices? I see other tuya ones that have a limit of 6.

closed 2023-04-02 by github-actions[bot]