Koenkk/zigbee2mqtt #16474: Neighbor device alwys join my network even if connecting are disabled.

Cosik, 2023-02-02:
### What happened?

Hi, I noticed that somehow my neighbor bulb is auto joining my zigbee network. I search google but didn't found such situation. I think that I've configure everything correctly and disable joining. 

Bulb model:
- Vendor: Philips, Model: 929001821618, Description: Hue white A60 bulb E27 bluetooth

Docker command
`docker run \
   --name=zigbee2mqtt \
   --restart=unless-stopped \
   --device=/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221201202339-if00:/dev/ttyACM0 \
   --network=host \
   -v /home/cosik_adm/homeassistant/zigbee2mqtt/data:/app/data \
   -v /run/udev:/run/udev:ro \
   -e TZ=Europe/Warsaw \
   --group-add dialout \
   koenkk/zigbee2mqtt`

Config file:
`homeassistant: true
permit_join: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.114
serial:
  port: /dev/ttyACM0
devices:
  '0xa4c138d092c49b43':
    friendly_name: kitchen_shutter
advanced:
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  channel: 15
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
device_options:
  legacy: false
`

### What did you expect to happen?

I expect that external devices will not join my network when I remove it and it is not allowed.

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 - Sonoff ZBDongle-E

### Debug log

[Zigbee2MQTT debug 2023.log](https://github.com/Koenkk/zigbee2mqtt/files/10567030/Zigbee2MQTT.debug.2023.log)
[image]

Gizzje, 2023-02-02:
Did you change all the default [network settings](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config), like the pan_id and network_key?

Cosik, 2023-02-02:
No I didn't change default values. My full config is in first post.

Gizzje, 2023-02-02:
I'm not an expert, but I can imagine that both you and your neighbour running a zigbee-network with the same settings can lead to this kind of behaviour. Do you know if your neighbour is also using Z2M?

Cosik, 2023-02-02:
Nope, he is not an a person who is using such solutions.

danieledwardgeorgehitchcock, 2023-02-03:
You could block / blacklist the device

closed 2023-03-13 by github-actions[bot]