Koenkk/zigbee2mqtt #16475: V1.30 broke Datex Apex smart plug (HLU2909K)

tjarmann, 2023-02-02:
### What happened?

After updating to 1.30 my three Datek Apex smart plugs have been listed at "unsupported" in zigbee2mqtt. 

I've tried to re-pair them, and the interview completes, but they are still listed as unsupported.

In the debug log the device I'm trying to pair is 'Switch04.Dryer'.

### What did you expect to happen?

I expected them to keep working like they did the day before

### How to reproduce it (minimal and precise)

Use a HLU2909K with zigbee2mqtt v1.30

### Zigbee2MQTT version

1.30

### Adapter firmware version

20220219

### Adapter

Sonoff ZBDongle-P CC2652P

### Debug log

warn  2023-02-02 10:33:57: Device 'Switch04.Dryer' left the network
info  2023-02-02 10:33:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Switch04.Dryer","ieee_address":"0x086bd7fffeb6b315"},"type":"device_leave"}'
debug 2023-02-02 10:33:57: Received Zigbee message from 'Dimmer1.KitchenTop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,90096099]}' from endpoint 1 with groupID 0
debug 2023-02-02 10:33:57: Received Zigbee message from 'Dimmer1.KitchenTop', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":1407}' from endpoint 1 with groupID 0
info  2023-02-02 10:33:57: MQTT publish: topic 'zigbee2mqtt/Dimmer1.KitchenTop', payload '{"brightness":254,"current":0.14,"energy":25.03,"last_seen":"2023-02-02T10:33:57+01:00","linkquality":116,"power":0.8,"state":"ON","voltage":140.8}'
info  2023-02-02 10:33:57: MQTT publish: topic 'zigbee2mqtt/Dimmer1.KitchenTop', payload '{"brightness":254,"current":0.14,"energy":25.03,"last_seen":"2023-02-02T10:33:57+01:00","linkquality":116,"power":0.8,"state":"ON","voltage":140.7}'
debug 2023-02-02 10:33:57: Received Zigbee message from 'Switch05.VVB', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":21}' from endpoint 1 with groupID 0
warn  2023-02-02 10:33:57: Received message from unsupported device with Zigbee model 'PoP' and manufacturer name 'Datek'
warn  2023-02-02 10:33:57: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2023-02-02 10:33:57: MQTT publish: topic 'zigbee2mqtt/Switch05.VVB', payload '{"current":9.23,"last_seen":"2023-02-02T10:33:57+01:00","linkquality":43,"power":2160,"state":"ON","temperature":41.41,"voltage":234.09}'
warn  2023-02-02 10:33:58: Device 'Switch04.Dryer' left the network
info  2023-02-02 10:33:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Switch04.Dryer","ieee_address":"0x086bd7fffeb6b315"},"type":"device_leave"}'
debug 2023-02-02 10:33:58: Received Zigbee message from 'Dimmer1.KitchenTop', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":10}' from endpoint 1 with groupID 0
info  2023-02-02 10:33:58: MQTT publish: topic 'zigbee2mqtt/Dimmer1.KitchenTop', payload '{"brightness":254,"current":0.14,"energy":25.03,"last_seen":"2023-02-02T10:33:58+01:00","linkquality":123,"power":1,"state":"ON","voltage":140.7}'
debug 2023-02-02 10:33:58: Received Zigbee message from 'Dimmer2.KitchenBot', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":1118}' from endpoint 1 with groupID 0
info  2023-02-02 10:33:58: MQTT publish: topic 'zigbee2mqtt/Dimmer2.KitchenBot', payload '{"brightness":254,"current":0.2,"energy":21.55,"last_seen":"2023-02-02T10:33:58+01:00","linkquality":69,"power":3,"state":"ON","voltage":111.8}'
debug 2023-02-02 10:33:59: Received Zigbee message from 'Dimmer3.Diningroom', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":71}' from endpoint 1 with groupID 0
info  2023-02-02 10:33:59: MQTT publish: topic 'zigbee2mqtt/Dimmer3.Diningroom', payload '{"brightness":254,"current":0.09,"energy":17.36,"last_seen":"2023-02-02T10:33:59+01:00","linkquality":112,"power":7.1,"state":"ON","voltage":137.8}'
debug 2023-02-02 10:33:59: Received Zigbee message from 'Dimmer3.Diningroom', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,62502758]}' from endpoint 1 with groupID 0
info  2023-02-02 10:33:59: MQTT publish: topic 'zigbee2mqtt/Dimmer3.Diningroom', payload '{"brightness":254,"current":0.09,"energy":17.36,"last_seen":"2023-02-02T10:33:59+01:00","linkquality":109,"power":7.1,"state":"ON","voltage":137.8}'
debug 2023-02-02 10:34:00: Received Zigbee message from 'Dimmer2.KitchenBot', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":30}' from endpoint 1 with groupID 0
info  2023-02-02 10:34:00: MQTT publish: topic 'zigbee2mqtt/Dimmer2.KitchenBot', payload '{"brightness":254,"current":0.2,"energy":21.55,"last_seen":"2023-02-02T10:34:00+01:00","linkquality":69,"power":3,"state":"ON","voltage":111.8}'
debug 2023-02-02 10:34:01: Received Zigbee message from 'Dimmer2.KitchenBot', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,77567011]}' from endpoint 1 with groupID 0
info  2023-02-02 10:34:01: MQTT publish: topic 'zigbee2mqtt/Dimmer2.KitchenBot', payload '{"brightness":254,"current":0.2,"energy":21.55,"last_seen":"2023-02-02T10:34:01+01:00","linkquality":69,"power":3,"state":"ON","voltage":111.8}'
debug 2023-02-02 10:34:02: Received Zigbee message from 'Dimmer1.KitchenTop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,90096105]}' from endpoint 1 with groupID 0
info  2023-02-02 10:34:02: MQTT publish: topic 'zigbee2mqtt/Dimmer1.KitchenTop', payload '{"brightness":254,"current":0.14,"energy":25.03,"last_seen":"2023-02-02T10:34:02+01:00","linkquality":120,"power":1,"state":"ON","voltage":140.7}'
info  2023-02-02 10:34:03: Accepting joining not in blocklist device '0x086bd7fffeb6b315'
info  2023-02-02 10:34:03: Device 'Switch04.Dryer' joined
info  2023-02-02 10:34:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Switch04.Dryer","ieee_address":"0x086bd7fffeb6b315"},"type":"device_joined"}'
info  2023-02-02 10:34:03: MQTT publish: topic 'zigbee2mqtt/Switch04.Dryer', payload '{"current":0.06,"last_seen":"2023-02-02T10:34:03+01:00","linkquality":58,"power":14,"state":"ON","temperature":33.45,"voltage":237.65}'
info  2023-02-02 10:34:03: Starting interview of 'Switch04.Dryer'
info  2023-02-02 10:34:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Switch04.Dryer","ieee_address":"0x086bd7fffeb6b315","status":"started"},"type":"device_interview"}'
info  2023-02-02 10:34:03: Accepting joining not in blocklist device '0x086bd7fffeb6b315'
debug 2023-02-02 10:34:03: Received Zigbee message from 'Dimmer1.KitchenTop', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":9}' from endpoint 1 with groupID 0
info  2023-02-02 10:34:04: MQTT publish: topic 'zigbee2mqtt/Switch04.Dryer', payload '{"current":0.06,"last_seen":"2023-02-02T10:34:03+01:00","linkquality":58,"power":14,"state":"ON","temperature":33.45,"voltage":237.65}'
info  2023-02-02 10:34:04: MQTT publish: topic 'zigbee2mqtt/Dimmer1.KitchenTop', payload '{"brightness":254,"current":0.14,"energy":25.03,"last_seen":"2023-02-02T10:34:03+01:00","linkquality":120,"power":0.9,"state":"ON","voltage":140.7}'
debug 2023-02-02 10:34:04: Received Zigbee message from 'Switch05.VVB', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":2,"rmsCurrent":11}' from endpoint 1 with groupID 0
warn  2023-02-02 10:34:04: Received message from unsupported device with Zigbee model 'PoP' and manufacturer name 'Datek'
warn  2023-02-02 10:34:04: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2023-02-02 10:34:04: MQTT publish: topic 'zigbee2mqtt/Switch05.VVB', payload '{"current":9.23,"last_seen":"2023-02-02T10:34:04+01:00","linkquality":43,"power":2160,"state":"ON","temperature":41.41,"voltage":234.09}'
[... 109 lines cut ...]
info  2023-02-02 10:34:25: MQTT publish: topic 'zigbee2mqtt/Switch04.Dryer', payload '{"current":0.06,"last_seen":"2023-02-02T10:34:25+01:00","linkquality":43,"power":14,"state":"ON","temperature":33.45,"voltage":237.65}'
debug 2023-02-02 10:34:25: Received Zigbee message from 'Switch04.Dryer', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2023-02-02 10:34:25: Skipping message, definition is undefined and still interviewing
[... cut ...]
debug 2023-02-02 10:34:26: Received Zigbee message from 'HAN01.BreakerBox', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltagePhC":2410}' from endpoint 1 with groupID 0
info  2023-02-02 10:34:26: MQTT publish: topic 'zigbee2mqtt/HAN01.BreakerBox', payload '{"current":7,"current_phase_b":65.54,"current_phase_c":10.4,"energy":121623.75,"last_seen":"2023-02-02T10:34:26+01:00","linkquality":112,"power":3083,"temperature":21,"update":{"installed_version":117531984,"latest_version":117531984,"state":"idle"},"update_available":null,"voltage":242.8,"voltage_phase_b":240.9,"voltage_phase_c":241}'

owangen, 2023-02-02:
A PR is already on it's way to fix the problem https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425. At least temporarily

Koenkk, 2023-02-02:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425 indeed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-02 by Koenkk