Koenkk/zigbee2mqtt #16477: wrong install code length

skrebber, 2023-02-02:
### What happened?

trying to pair a Bosch TwinGuard smoke detector, as a Zigbee 3.0 device it needs an install code. The QR code of from device is:
`RB01SG0D836591B3CC0010000000000000000000000D6F0018869DDBDLK82AF21EF5510E04E0459F250813F9562`

z2m says, wrong install code length. 

Interesting:
I first used 1.29.2 bbbaaf4 with the same experience. Then updated to 15.0.39 ea0a999 with the same negative result. But @safakaltun tested my install code in his envoirment with success. see: https://github.com/Koenkk/zigbee2mqtt/issues/14355#issuecomment-1412790420 

### What did you expect to happen?

successful paring of the Zigbee 3.0 device.

### How to reproduce it (minimal and precise)

press -Settings-Tools-Add install code paste code from above an click ok.

### Zigbee2MQTT version

15.0.39

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant SkyConnect

### Debug log

```
Debug 2023-02-02 10:34:57Received MQTT message on 'zigbee2mqtt/bridge/request/install_code/add' with data '{"transaction":"7m2tf-1","value":"RB01SG0D836591B3CC0010000000000000000000000D6F0018869DDBDLK82AF21EF5510E04E0459F250813F9562"}'
Error 2023-02-02 10:34:57Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Wrong install code length'
Debug 2023-02-02 10:34:58Error: Wrong install code length at EZSPAdapter.addInstallCode (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:260:19) at Controller.addInstallCode (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:230:28) at Zigbee.addInstallCode (/app/lib/zigbee.ts:314:29) at Bridge.installCodeAdd (/app/lib/extension/bridge.ts:229:27) at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:122:63) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28)
Info 2023-02-02 10:34:58MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Wrong install code length","status":"error","transaction":"7m2tf-1"}'
```

dan3o3, 2023-02-13:
Experiencing the exact same problem with a Bosch Twinguard. Also a fillup with 4 zeros before "DLK" let Z2M accept the code but it will not work as the device will not pair. I successfully paired a Motion Controller (Bosch) and a Plug 2 successfully, so it seems to be something special with these Twinguard codes...

l154, 2023-03-16:
Hi I have exactly the same error when I'm trying connect to Bosh Smart Home Twinguard.
 2023-03-16 09:55:10Received MQTT message on 'zigbee2mqtt/bridge/request/install_code/add' with data '{"transaction":"kge09-1","value":"RB01SG0D836591B3CC0010000000000000000000000D6F0017E1B786DLK0CEC411FCB2F415FB95ED8E4C9C4F316"}' Error 2023-03-16 09:55:10Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Wrong install code length' Debug 2023-03-16 09:55:10Error: Wrong install code length at EZSPAdapter.addInstallCode (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:260:19) at Controller.addInstallCode (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:230:28) at Zigbee.addInstallCode (/app/lib/zigbee.ts:314:29) at Bridge.installCodeAdd (/app/lib/extension/bridge.ts:229:27) at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:122:63) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28) Info 2023-03-16 09:55:10MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Wrong install code length","status":"error","transaction":"kge09-1"}'

Swisyn, 2023-03-24:
I also have the same issue, even though the pairing code length on QR is 91 - I still get wrong code length alert. 

What could be the main reason of this issue? 

`RB01SG0D836591B3CC0010000000000000000000000D6F0018860CDCDLKE8F6BCB11962F4377573AEB18480CF7E`

OnkelMikke, 2023-04-01:
hi cross cheked with me devices.
first smart plug II
QR: RB01SG0D83101826480030000000000000000000003C84FFFEEDBE4CDLK5D52BEA89C2E0758FD2FEBE1FE3932C2289F
IEEE: 00:3C:84:FF:FE:ED:BE:4C
Install-Code: 5D52-BEA8-9C2E-0758-FD2F-EBE1-FE39-32C2-289F (printed at label, match with qr and works)

Looks like, IEEE is bevor DLK and after DLK Install Code. 

checked Twinguard QR, looks the same.
RB01SG0D836591B3CC0010000000000000000000000D6F000F4F53B8DLK2E96368640299BAB76865A0893E1F735
IEEE: 00:0D:6F:00:0F:4F:53:88 (printed at back label)
Thatsway i think my install code should be 
2E96-3686-4029-9BAB-7686-5A08-93E1-F735

Not checked yet. I test it and comment later, again.

closed 2023-05-09 by github-actions[bot]

NicolaiSoeborg, 2023-07-07:
In [release 1.32.1](https://github.com/Koenkk/zigbee2mqtt/pull/18182) the length issue has been fixed, but I still have some issues with the next ("interview") step during pairing, just a FYI.

4s1, 2023-12-03:
Did you get it to work @NicolaiSoeborg? 

I'd guess by comments on #14355 this could be an coordinator problem?
Looks like it works with CC2652R coordinators. But I'm not able to pair on my HA yellow (ezsp).