Koenkk/zigbee2mqtt #16479: Xiaomi ZNCLBL01LM: Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'

buhito81, 2023-02-02:
### What happened?

Very frequent error

```
Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'
Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{% if not value_json.motor_state %} stopped {% else %} {{ value_json.motor_state }} {% endif %}'
```
I've tried, with reinstalls, empty databases, emptying MQTT database/cache, but the situation remains the same: motor_state appears to be expected constantly, but is not communicated constantly.

Devices have been re-paired etc.

[image]

Previously reported in the following issue:

https://github.com/Koenkk/zigbee2mqtt/issues/15116

### What did you expect to happen?

No errors

### How to reproduce it (minimal and precise)

Connect the ZNCLBL01LM device to Zigbee2MQTT, have it connected for a day.

### Zigbee2MQTT version

1.27.2 commit: unknown, 1.28.2 commit: unknown, 1.30.0-1 commit: unknown

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

2022-11-21 10:22:28.624 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on zigbee2mqtt/Bedroom Curtain Right: b'{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}'
2022-11-21 10:22:28.625 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_battery', 'name': 'Bedroom Curtain Right battery', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_battery)>} with default value '100' and Template("{{ value_json.battery }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_device_temperature', 'name': 'Bedroom Curtain Right device temperature', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_device_temperature)>} with default value '18' and Template("{{ value_json.device_temperature }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_action', 'name': 'Bedroom Curtain Right action', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_action)>} with default value 'None' and Template("{{ value_json.action }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_motor_state', 'name': 'Bedroom Curtain Right motor state', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_motor_state)>} with default value '' and Template("{{ value_json.motor_state }}")
2022-11-21 10:22:28.626 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'
2022-11-21 10:22:28.629 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_hooks_state', 'name': 'Bedroom Curtain Right hooks state', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_hooks_state)>} with default value 'locked' and Template("{{ value_json.hooks_state }}")
2022-11-21 10:22:28.630 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_target_position', 'name': 'Bedroom Curtain Right target position', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_target_position)>} with default value '72' and Template("{{ value_json.target_position }}")
2022-11-21 10:22:28.630 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_power_source', 'name': 'Bedroom Curtain Right power source', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_power_source)>} with default value 'battery' and Template("{{ value_json.power_source }}")
2022-11-21 10:22:28.630 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'binary_sensor.bedroom_curtain_right_running', 'name': 'Bedroom Curtain Right running', 'this': <template TemplateStateFromEntityId(binary_sensor.bedroom_curtain_right_running)>} and Template("{{ value_json.running }}")
2022-11-21 10:22:28.631 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'binary_sensor.bedroom_curtain_right_charging', 'name': 'Bedroom Curtain Right charging', 'this': <template TemplateStateFromEntityId(binary_sensor.bedroom_curtain_right_charging)>} and Template("{{ value_json.charging }}")
2022-11-21 10:22:28.631 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source" [... cut ...] m_curtain_right', 'name': 'Bedroom Curtain Right', 'this': <template TemplateStateFromEntityId(cover.bedroom_curtain_right)>} and Template("{% if not value_json.motor_state %} stopped {% else %} {{ value_json.motor_state }} {% endif %}")
2022-11-21 10:22:28.631 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{% if not value_json.motor_state %} stopped {% else %} {{ value_json.motor_state }} {% endif %}'

buhito81, 2023-02-02:
As I understand https://github.com/Koenkk/zigbee2mqtt/pull/16017 or actually https://github.com/Koenkk/zigbee2mqtt/pull/16324 therein still not resolve the motor_state not always being there, since well before and (now) after these, this remains an issue.

Koenkk, 2023-02-12:
Should be fixed by https://github.com/Koenkk/zigbee2mqtt/pull/16614 thanks to @Drafteed 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-12 by Koenkk

buhito81, 2023-03-27:
Can't comment anymore under https://github.com/Koenkk/zigbee2mqtt/pull/16614 as requested, but here are the files:

[mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Livingroom Curtain Left-6922d96b184a49b1f063241f497cbf81.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/11078705/mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Livingroom.Curtain.Left-6922d96b184a49b1f063241f497cbf81.json.txt)
[mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Bedroom Curtain Right-140c74b126b79912ff4e4bc64d007257.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/11078706/mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Bedroom.Curtain.Right-140c74b126b79912ff4e4bc64d007257.json.txt)
[mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Bedroom Curtain Left-94add9909859213dc4cb1e85cb8d417d.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/11078707/mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Bedroom.Curtain.Left-94add9909859213dc4cb1e85cb8d417d.json.txt)
[mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Livingroom Curtain Right-8d58335a3ac5bdb75839219f82b85d3e.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/11078709/mqtt-bb7a7f55158f036ddd32fc20ad8636eb-Livingroom.Curtain.Right-8d58335a3ac5bdb75839219f82b85d3e.json.txt)

PR #16614 (2023-02-12): Improve checks in HA covers
- Add check if `motor_state` is not exists (Fix #16479);
- Add check if `running` is not exists;
- Use found `stoppedState`;
- Reorder to use fallback if lookup fails.