Koenkk/zigbee2mqtt #16481: ERROR: Failed to execute LQI for 'Coordinator'

kernel85, 2023-02-02:
### What happened?

Hi everyone,

I'm experiencing, time by time, an annoying issue with my EZSP Coordinator `ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812084548-if00` which stops receiving data from end devices. 
When accessing the "Map" feature, from the Frontend, I get the following error:

`ERROR: Failed to execute LQI for 'Coordinator'` and no devices are connected to the Coordinator. 

Only restarting the Z2M add-on brings the situation back to normal, but after a couple of hours or days (I cannot correlate why), it goes down again. 

Before opening this issue I read different threads regarding this particular issue and tried to apply some advice that could have helped to solve it definitely but nothing has changed so far. I've tried to:

* switch to the Z2M Edge version `1.29.2-dev` commit: `[6ce2d2e](https://github.com/Koenkk/zigbee2mqtt/commit/6ce2d2e)`
* disabled the "Availability" feature; 

Right now, I don't know how to proceed to fix it. I could provide help in troubleshooting the issue. Additionally, I can provide any information that could be useful for debugging purposes. 

### What did you expect to happen?

To have devices properly connected to the coordinator w/o unexpected disconnections. 

### Zigbee2MQTT version

1.29.2-dev

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2 20220812084548-if00

### Debug log

`2023-02-02T08:39:11.981Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2023-02-02T08:39:11.982Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4e80010500
2023-02-02T08:39:11.983Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2023-02-02T08:39:11.983Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2023-02-02T08:39:20.263Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-02-02T08:39:20.265Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-02-02T08:39:20.266Z zigbee-herdsman:adapter:ezsp:driv Close driver
Zigbee2MQTT:debug 2023-02-02 09:40:38: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 09:45:38: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 09:50:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 09:55:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:00:39: Saving state to file /config/zigbee2mqtt/state.json
2023-02-02T09:00:43.426Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
Zigbee2MQTT:debug 2023-02-02 10:05:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:10:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:15:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:20:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:25:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:30:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:35:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:40:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:45:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:50:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 10:55:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:00:39: Saving state to file /config/zigbee2mqtt/state.json
2023-02-02T10:00:43.428Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
Zigbee2MQTT:debug 2023-02-02 11:05:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:10:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:15:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:20:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:25:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:30:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:35:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 11:40:39: Saving state to file /config/zigbee2mqtt/state.json
[... 41 lines cut ...]
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023-02-02T13:04:25.142Z zigbee-herdsman:adapter:ezsp:erro zdoRequest error
2023-02-02T13:04:30.149Z zigbee-herdsman:adapter:ezsp:driv ZDO Mgmt_Lqi_req params: {"startindex":0}
2023-02-02T13:04:30.150Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[224,121,141,255,254,113,148,172]}},"extendedTimeout":true}
2023-02-02T13:04:30.160Z zigbee-herdsman:adapter:ezsp:erro Request error Error: Connection not initialized: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:505:19)
    at Driver.request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:473:29)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:567:21)
    at request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:275:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:300:28)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023-02-02T13:04:30.160Z zigbee-herdsman:adapter:ezsp:erro zdoRequest error
Zigbee2MQTT:error 2023-02-02 14:04:30: Failed to execute LQI for 'Coordinator'
Zigbee2MQTT:debug 2023-02-02 14:04:30: Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:571:19)
    at request (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:275:32)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:300:28)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2023-02-02 14:04:30: Network scan finished
Zigbee2MQTT:info  2023-02-02 14:04:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[],"nodes":[{"definition":null,"failed":["lqi"],"friendlyName":"Coordinator","ieeeAddr":"0xe0798dfffe7194ac","lastSeen":1674910271429,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Temperature and humidity sensor","model":"SNZB-02","supports":"battery, temperature, humidity, voltage, linkquality","vendor":"SONOFF"},"friendlyName":"Sensore Temperatura Bagno","ieeeAddr":"0x00124b0029202ae4","lastSeen":1675327142424,"manufacturerName":"eWeLink","modelID":"TH01","networkAddress":30056,"type":"EndDevice"},{"definition":{"description":"Temperature and humidity sensor","model":"SNZB-02","supports":"battery, temperature, humidity, voltage, linkquality","vendor":"SONOFF"},"friendlyName":"Sensore Temperatura Cabina Armadio","ieeeAddr":"0x00124b0029202c78","lastSeen":1675326870668,"manufacturerName":"eWeLink","modelID":"TH01","networkAddress":47101,"type":"EndDevice"},{"definition":{"description":"Aqara temperature, humidity and pressure sensor","model":"WSDCGQ11LM","supports":"battery, temperature, humidity, pressure, voltage, linkquality","vendor":"Xiaomi"},"friendlyName":"Sensore Temperatura Cucina","ieeeAddr":"0x00158d0008fff081","lastSeen":1675325664063,"manufacturerName":"LUMI","modelID":"lumi.weather","networkAddress":9158,"type":"EndDevice"},{"definition":{"description":"Aqara temperature, humidity and pressure sensor","model":"WSDCGQ11LM","supports":"battery, temperature, humidity, pressure, voltage, linkquality","vendor":"Xiaomi"},"friendlyName":"Sensore Temperatura Cameretta","ieeeAddr":"0x00158d0008ffef0b","lastSeen":1675326478305,"manufacturerName":"LUMI","modelID":"lumi.weather","networkAddress":25218,"type":"EndDevice"},{"definition":{"description":"Aqara temperature, humidity and pressure sensor","model":"WSDCGQ11LM","supports":"battery, temperature, humidity, pressure, voltage, linkquality","vendor":"Xiaomi"},"friendlyName":"Sensore Temperatura Camera","ieeeAddr":"0x00158d0008ffeef5","lastSeen":16753263447 [... cut ...] }]}},"status":"ok","transaction":"tp7kf-1"}'
Zigbee2MQTT:debug 2023-02-02 14:05:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 14:10:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 14:15:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 14:20:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-02-02 14:25:39: Saving state to file /config/zigbee2mqtt/state.json`

kirovilya, 2023-02-02:
why are you not using the latest version of z2m 1.30.0? try it on latest version

kernel85, 2023-02-02:
I'll give it a try and let you know if anything changes.

closed 2023-03-12 by github-actions[bot]

tommaineult, 2023-09-20:
Hi @kernel85 , Has this problem been solved by version of z2m 1.30.0?

kernel85, 2023-09-20:
Hi @tommaineult,

yes, I can confirm the issue has been solved by upgrading to v1.30.0.

bricololo44, 2024-07-19:
I have a slzb-06p7, when i connected the sonoff router, i obtained a default lqi on coordinator.
Need to restor a last image without sonoff router