Koenkk/zigbee2mqtt #16488: Toggling HK-DIM-A on/off results in timeouts. Changing brightness doesn't

Waseh, 2023-02-02:
### What happened?

I'm having issues with a couple of devices that zigbee2mqtt recognizes as Candeo HK-DIM-A dimmer switches but are branded as [Light Solutions](https://ledpaerer.dk/zigbee-dreje-lysdaemper-til-lk-250w.html) dimmer switches.

Pairing is painless but switching the switches on and off from zigbee2mqtt results in timeouts, but does eventually turn on and off the switch. However if only setting the brightness value no timeouts happen and the switches are fast and responsive. Controlling the switch only by setting the brightness value (0 for off and >0 for on) works perfectly fine
The state is correctly updated in zigbee2mqtt when using the physical switch to toggle and updates very fast.

The same behavior is seen on both devices.

I've updated to the newest version of zigbee2mqtt as well as flashed the newest firmware on my coordinator with the same results. Re-pairing didn't help either.

### What did you expect to happen?

Toggling on and off to work without timing out and being slow. Expected to work as fast and reliable as changing brightness.

### How to reproduce it (minimal and precise)

Turn device on or off via zigbee2mqtt.

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10571403/log.txt)

Waseh, 2023-02-06:
As additional information if i pair the device to my old IKEA Tradfri hub the device works as expected and responds without delay to on/off commands

bjornbak, 2023-02-10:
@Waseh I experience the same issue with a [Hilux Zigbee DIN dimmer](https://ledpaerer.dk/zigbee-din-rail-dimmer.html) which is apparently the same device internally. 

I've tested with both Zigbee2mqtt and ZHA. Zigbee2mqtt will recover from a timeout and after awhile find the correct state where as ZHA is frozen in the wrong state until I use a physical button connected to the dimmer..

Waseh, 2023-02-10:
@bjornbak I'm using a workaround where I've paired the offending devices to my old Trådfri hub (ironically one i was seeking to replace with zigbee2mqtt) and using the Ikea integration in Home Assistant to control them.
This works flawlessly, both on/off and brightness changes without delay, but kinda defeats the purpose.

I'm glad i didn't go back to try ZHA since you've had even less success there.

What's even more annoying is i have 3 older models of the same switch that work fine in zigbee2mqtt, but they identify as Eco-Dim.07/Eco-Dim.10

bjornbak, 2023-02-10:
@Waseh I went out and bought a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-**E** which have another chipset. 
My hope is this will have similar effect to using the IKEA hub. :crossed_fingers:
Will let you know once I get it tested.

Waseh, 2023-02-10:
Please keep me posted! Hope it works

bjornbak, 2023-02-11:
@Waseh No luck :(
The situation is just as bad with Dongle-E and ZHA..
I'll see if I can reproduce your good behavior using the IKEA hub.

Bahodirbm, 2023-02-11:
Hi, just created account to answer my solution 😀. In z2mqtt setting (specific) set transition to 0 and set to state action true. It worked for me. I have exactly the same dimmer.

Bahodirbm, 2023-02-11:
@Waseh @bjornbak . FYI

bjornbak, 2023-02-11:
@Bahodirbm Thank you so much for your effort! :) 
Is this a global z2m setting or just for the dimmer? Maybe share a screenshot?
Would like to see if I can make it work in ZHA before migrating back to z2m...

Bahodirbm, 2023-02-11:
It was under dimmer's settings. When I had it with ZHA it worked fine for me. But then decided to move whole setup to mqtt and i had issue with this dimmer.

Bahodirbm, 2023-02-11:
[image]

Bahodirbm, 2023-02-11:
@bjornbak

closed 2023-03-22 by github-actions[bot]

esbenr, 2023-09-18:
Just bought the same device with firmware v. 20220523. Same issue. The fix above fixes the timeouts, but not the error when trying to set the power-on-state. 

Publish 'set' 'power_on_behavior' to '0x385b44fffe1b3aa8' failed: 'Error: Write 0x385b44fffe1b3aa8/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')

The Odd part is that i bought the same dimmer  one year ago with the firmware v. 20200701 and that works without quirks. That registreres as “Dimmer-Switch-ZB3.0” whereabouts the faulty version registers as HK-DIM-A. 

Its obvious that LightSolutions switched vendor and are using another chipset, that has compatibility issues with Z2M.

simplenotezy, 2024-01-31:
I am not sure if this is related to the issues described by @esbenr and @Waseh - we have the same dimmers running on Zigbee with Conbee II.

We notice that we have issues when turning off the device via HA, it turns off the lights, but an error will occur after a few seconds saying "Failed to turn off. Device did not respond".

We also notice that when you manually toggle the power state on/off on the physical switch, it does not report anything to the logbook.

CGraabaek, 2024-11-17:
@bjornbak did you ever find a solution to this? I have just bought a house that have this dimmer, and I wanted to switch it out from the "Smart ME" by Martimek. 

After pairing it, i see the same symptoms as you do :(

bjornbak, 2024-11-18:
@CGraabaek I don't remember how I fixed it, but they are working correctly in my house with Z2M 1.41.0 and the settings @Bahodirbm suggested above.
More recently I contributed a Herdsman converter for another variant of the Hilux Dimmer: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7763.
So I believe they work out-of-the-box with at most you need to adjust the settings :)

Be aware that Dimmer have a Min button on the front setting the min watts the dimmer can be adjusted too. Many bulbs require a relatively high amount of watts before they start to lighten up. If you bulbs eg require 50W you can turn on/off without it working notable if the Dimmer is set to 40W...