Koenkk/zigbee2mqtt #16490: Zigbee2MQTT fails to start : Segmentation fault

Patrice7, 2023-02-02:
### What happened?

I have installed v1.24 on Openwrt, start without problem 
I have installed actual v1.30.0 with this error :

> Zigbee2MQTT:info  2023-02-02 19:01:06: Starting Zigbee2MQTT version 1.30.0 (commit #8f781db)
> Zigbee2MQTT:info  2023-02-02 19:01:06: Starting zigbee-herdsman (0.14.89)
> Segmentation fault

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20210708

### Adapter

Sonoff cc2531

Patrice7, 2023-02-04:
v1.28.4 is ok
v1.29.0 and high same error

roth-m, 2023-02-04:
Same here!
I suspect the issue is with module serialport...

roth-m, 2023-02-04:
Success, downgraded to zigbee-herdsman (0.14.76) and it is now lauching on OpenWRT.

See the changes applied on serialport between the 2 versions here:

https://github.com/Koenkk/zigbee-herdsman/compare/v0.14.76...v0.14.89

Koenkk, 2023-02-05:
Try executing `npm rebuild --build-from-source` in the z2m directory.

Patrice7, 2023-02-05:
With 'npm rebuild --build-from-source' is ok, thanks

closed 2023-02-05 by Patrice7

nragon, 2023-02-16:
I can confirm that this happens
uname -a: Linux FriendlyWrt 5.15.78
z2m version: 1.30.1
node --version: v16.19.0
npm --version: 8.19.3
node-serialport: 9.0.4-2
node-serialport-bindings: 9.0.4-2