Koenkk/zigbee2mqtt #16491: Datek wallplug stopped working

damtjern, 2023-02-02:
### What happened?

After update from 1.29 home assistant addon to 1.30.0 my wall switches became "unsupported" and stops working.
I have 3 of them all came up unavailable.
Should be of this type, not sure if there is multiple revisions but they have not showed up in OTA as stated they support.
https://www.zigbee2mqtt.io/devices/HLU2909K.html

### Zigbee2MQTT version

1.30.0-1

### Adapter firmware version

20220219

### Adapter

Electrolama zig-ah-zag-ah!

### Debug log

13:21Received Zigbee message from 'office_floorlamp', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":24661}' from endpoint 1 with groupID 0
Warning 2023-02-02 19:13:21Received message from unsupported device with Zigbee model 'PoP' and manufacturer name 'Datek'
Warning 2023-02-02 19:13:21Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html``

korsmann, 2023-02-02:
Same problem, reports as unsupported.

svhelge, 2023-02-02:
Fixed with https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425

atus, 2023-02-02:
Confirming issue. Rollback to previous version fixes it.

Koenkk, 2023-02-04:
Will release a hotfix release in the coming days.

closed 2023-02-04 by Koenkk