Koenkk/zigbee2mqtt #16500: Unsupported device with Zigbee model 'ID Lock 150' and manufacturer name 'Datek'

davidkson, 2023-02-03:
### What happened?

After the latest release 1.30.0-1 Zigbee model 'ID Lock 150' shows as unsupported and it's not possible to communicate with the lock through Home Assistant

### What did you expect to happen?

To work as before

### How to reproduce it (minimal and precise)

Install 1.30.0-1

### Zigbee2MQTT version

1.30.0-1

### Adapter firmware version

20211217

### Adapter

Sonoff 3.0

thrawnarn, 2023-02-04:
I have the same problem :(

Koenkk, 2023-02-04:
Will be fixed in tomorrows hotfix release.

closed 2023-02-04 by Koenkk

odinr, 2023-02-06:
I have the same issue, can't find any hotfix or commit for this issue

davidkson, 2023-02-06:
Iam running the "Edge"-version right now and then the ID Lock works as before