Koenkk/zigbee2mqtt #16502: Add install code is not supported

Sailing74, 2023-02-03:
### What happened?

In order to pair a Bosch BTH-RA, I am trying to set the install code. Setting the install code scanned from the QR code, I am getting:

error 2023-02-03 11:36:42: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Add install code is not supported'
info  2023-02-03 11:36:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Add install code is not supported","status":"error","transaction":"tazyy-1"}'

(I Xed out the install code in the debug log, as I don't know if it's safe to share that publicly. If you need it for debug purposes and it should be safe to share it publicly, please let me know.)

### What did you expect to happen?

Store install code, so that I can pair the device

### Zigbee2MQTT version

1.30

### Adapter firmware version

0x26720700

### Adapter

ConBee II

### Debug log

debug 2023-02-03 11:44:14: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2023-02-03 11:44:14: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-02-03.11-44-14' filename: log.txt
info  2023-02-03 11:44:14: Starting Zigbee2MQTT version 1.30.0 (commit #8f781db)
info  2023-02-03 11:44:14: Starting zigbee-herdsman (0.14.89)
debug 2023-02-03 11:44:14: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
info  2023-02-03 11:44:15: zigbee-herdsman started (resumed)
info  2023-02-03 11:44:15: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
debug 2023-02-03 11:44:15: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2023-02-03 11:44:15: Currently 0 devices are joined:
warn  2023-02-03 11:44:15: `permit_join` set to  `true` in configuration.yaml.
warn  2023-02-03 11:44:15: Allowing new devices to join.
warn  2023-02-03 11:44:15: Set `permit_join` to `false` once you joined all devices.
info  2023-02-03 11:44:15: Zigbee: allowing new devices to join.
info  2023-02-03 11:44:16: Connecting to MQTT server at mqtt://192.168.1.107:1883
debug 2023-02-03 11:44:16: Using MQTT login with username: xxx
debug 2023-02-03 11:44:16: Using MQTT client ID: 'xxx'
info  2023-02-03 11:44:16: Connected to MQTT server
info  2023-02-03 11:44:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2023-02-03 11:44:16: Started frontend on port 0.0.0.0:8080
debug 2023-02-03 11:44:16: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"8f781db","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.30.0"}'
info  2023-02-03 11:44:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"8f781db","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.30.0"}'
info  2023-02-03 11:44:16: Zigbee2MQTT started!
debug 2023-02-03 11:44:46: Received MQTT message on 'zigbee2mqtt/bridge/request/install_code/add' with data '{"transaction":"yioth-1","value":"RB01SG0D83101826480080000000000000000000XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX4"}'
error 2023-02-03 11:44:46: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Add install code is not supported'
debug 2023-02-03 11:44:46: Error: Add install code is not supported
    at DeconzAdapter.addInstallCode (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:188:31)
    at Controller.addInstallCode (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:230:28)
    at Zigbee.addInstallCode (/opt/zigbee2mqtt/lib/zigbee.ts:314:29)
    at Bridge.installCodeAdd (/opt/zigbee2mqtt/lib/extension/bridge.ts:229:27)
    at Bridge.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/bridge.ts:122:63)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:140:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:121:27)
    at WebSocket.emit (node:events:513:28)
info  2023-02-03 11:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Add install code is not supported","status":"error","transaction":"yioth-1"}'
debug 2023-02-03 11:45:05: Saving state to file /opt/zigbee2mqtt/data/state.json
info  2023-02-03 11:45:05: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2023-02-03 11:45:05: Disconnecting from MQTT server
info  2023-02-03 11:45:05: Stopping zigbee-herdsman...
info  2023-02-03 11:45:06: Stopped zigbee-herdsman
info  2023-02-03 11:45:06: Stopped Zigbee2MQTT

skrebber, 2023-02-03:
maybe related to issue https://github.com/Koenkk/zigbee2mqtt/issues/16477

Sailing74, 2023-02-04:
Interestingly, I‘m having the same problem with a Bosch BSP-FZ2: Scanning the install code and submitting it in the frontend leads to the same error. I‘m probably doing something wrong, but what is it?

Sailing74, 2023-02-06:
Ok, so not sure if I missed the hint somewhere. It seems that "install code" is not supported with every coordinator. If I exchange the conbee ii with a sonoff zigbee stick, install code works.

closed 2023-03-16 by github-actions[bot]