Koenkk/zigbee2mqtt #16504: QOXEZY Zigbee CO2 NDIR Sensor

OlsiIgitt, 2023-02-03:
### Link

https://www.amazon.de/NDIR-Kohlendioxid-Luftqualit%C3%A4ts-Alarmeinstellungen-Umgebungstemperatur-Luftfeuchtigkeit/dp/B0BN5YW1RZ/ref=sr_1_1?keywords=QOXEZY+ZIGBEE+CO2+NDIR&link_code=qs&qid=1675361945&sourceid=Mozilla-search&sr=8-1

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0xa4c138309c069fb2","nwkAddr":24399,"manufId":4417,"manufName":"_TZE200_3ejwxpmu","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i´┐¢´┐¢n+i","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_3ejwxpmu","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1675418155595,"defaultSendRequestWhen":"immediate"}

### Comments

This is a NDIR CO2 sensor which measures carbon dioxide (additionally humidity and temperature; Tuya device?); see picture:
[image]
With investigation and many reading (and luck) I had success to connect to this new device, which is not supported by Zigbee2MQTT at this day. I changed the file tuya.js which I found in path ..\node_modules\zigbee-herdsman-converters\devices\. I added the following block:

    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_3ejwxpmu'}],
        model: 'TS0601_CO2_NDIR_Sensor',
        vendor: 'TuYa',
        description: 'CO2 NDIR Sensor',
        fromZigbee: [fz.tuya_air_quality],
        toZigbee: [],
        exposes: [e.temperature(), e.humidity(), e.co2()],
    },

I don't know whether this is the correct and "legal" way, but nevertheless I had succes with this modification: Now I can read all measuring values CO2, Humidity, and Temperature!!! BTW: The sensor will update its measuring values every ~ second (exactly 825 ms).

### External converter

```shell
???
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_3ejwxpmu'}],
        model: 'TS0601_CO2_NDIR_Sensor',
        vendor: 'TuYa',
        description: 'CO2 NDIR Sensor',
        fromZigbee: [fz.tuya_air_quality],
        toZigbee: [],
        exposes: [e.temperature(), e.humidity(), e.co2()],
    }
???
```

Koenkk, 2023-02-04:
Can you try with the following `exposes`: `exposes: [e.temperature(), e.humidity(), e.co2(), e.voc().withUnit('ppm'), e.formaldehyd()],` and see if it also published those values? (`voc`/`formaldehyd`)

OlsiIgitt, 2023-02-04:
> Can you try with the following exposes: exposes: [e.temperature(), e.humidity(), e.co2(), e.voc().withUnit('ppm'), e.formaldehyd()], and see if it also published those values? (voc/formaldehyd)

No, i does not! I receive the same response as for 
"exposes: [e.temperature(), e.humidity(), e.co2()]"
I would also have been surprised, since the product is advertised with CO2, temperature and humidity...

OlsiIgitt, 2023-02-04:
I found another distributor link for this device:
https://www.amazon.de/Detector-Infrared-Temperature-Humidity-Detection/dp/B0BP7ZSDFH/ref=zg_bs_14261491031_sccl_16/258-5334715-5027056?psc=1

 I think it's the same product. Might be that the manufaturer is Tuya, but I am not sure.  Anyway, I can't find this CO2 NDIR analyzer on Tuya web pages.

Koenkk, 2023-02-05:
Thanks, I've added support for this device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-05 by Koenkk