Koenkk/zigbee2mqtt #16507: [New device support]: high power relay  - new device or not?

rsteininger, 2023-02-03:
### Link

https://www.aliexpress.com/item/1005005037224711.html

### Database entry

{"id":61,"type":"Router","ieeeAddr":"0xa4c1388fbc99f99f","nwkAddr":28709,"manufId":4417,"manufName":"_TZ3000_tgddllx4","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ï¿½\u0003\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_tgddllx4","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":4,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,11]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ffe5c","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1675433792657,"defaultSendRequestWhen":"immediate"}

### Comments

I recently bought a high amp relay switch that was advertised with energy monitoring. 

COLOROCK CR-MNZ1 30A relay with power monitoring
https://www.aliexpress.com/item/1005005037224711.html

However, zigbee2mqtt finds it to be a "know" switch without power monitoring: 
Zigbee Model: TS0001
Zigbee Manufacturer: _TZ3000_tgddllx4
Description: 1 gang switch
Support status: Supported

The database entry does state: "haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,11]}}

So, I assume, this wasn't "false advertising" but simply not recognised as a device that does have power monitoring as it unfortunately shares a device name with one that does not ... I don't have any other devices with Model: TS0001!

So my ask is, how can this be "fixed" ... any suggestion? And will this "fix" work after the next zigbee2mqtt version update? 

THANKS!

Koenkk, 2023-02-04:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-04 by Koenkk

alandrumo, 2024-04-20:
I bought a Cr-mnz1 and it was delivered today. It appears to be without energy monitoring and cr- mn2 is with the monitoring. When i added it to home assistant power and voltage etc were listed but stayed at 0

jacasal, 2024-07-07:
I´m in the same situation.

[image]
[image]

lpt2007, 2024-07-18:
I have same problem @jacasal, power monitoring shows N/A. Any idea?

rheos13, 2024-08-11:
I have the same problem as @jacasal and @lpt2007

Zigbee Model: TS0001
Zigbee Manufacturer: _TZ3000_tgddllx4
Description: 1 gang switch  30A with power monitoring
Support status: Supported
Manufacturer: Colorock
Model: CR-MNZ1

Zigbee2MQTT version 1.39.1

ERROR:
[2024-08-11 16:00:02] error: 	z2m: Failed to configure '0x70b3d52b60092cf3b3d', attempt 1 (Error: ZCL command 0x70b3d52b60092cf3b3d/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (C:\Zigbee2MQTT\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:351:28)
    at Endpoint.zclCommand (C:\Zigbee2MQTT\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:870:26)
    at Endpoint.configureReporting (C:\Zigbee2MQTT\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:611:9)
    at Object.rmsVoltage (C:\Zigbee2MQTT\node_modules\zigbee-herdsman-converters\src\lib\reporting.ts:221:5)
    at Object.configure (C:\Zigbee2MQTT\node_modules\zigbee-herdsman-converters\src\devices\tuya.ts:3392:13)
    at Configure.configure (C:\Zigbee2MQTT\lib\extension\configure.ts:126:13)
    at EventEmitter.wrappedCallback (C:\Zigbee2MQTT\lib\eventBus.ts:174:17))

rheos13, 2024-08-11:
Hi again,

 With this external converter, it works! [https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa](url)

It seems that it is not included in the version 1.39.1

regards!