Koenkk/zigbee2mqtt #16511: Gledopto LED controllers keep spaming with OTA request - no response?

chojnicki, 2023-02-03:
### What happened?

Similar to https://github.com/Koenkk/zigbee2mqtt/issues/16371 but with Gledopto [GL-C-009P](https://www.zigbee2mqtt.io/devices/GL-C-009P.html#gledopto-gl-c-009p) and [GL-C-008P](https://www.zigbee2mqtt.io/devices/GL-C-008P.html#gledopto-gl-c-008p).
I noticed in logs that every single minute it does:

`Device 'living_room_cabinet_light' requested OTA` and nothing else about this device until next minute.
When I check OTA manually in dashboard I have two situations:

1. For GL-C-008P - `No update available for 'entrance_ceiling_lamp'`
2. For GL-C-009P - `Failed to check if update available for 'living_room_cabinet_light' (No image available for imageType '5142')`

Second one is error, but in case 008P if there is simply no update available, shouldn't it respond to device with "no update, leave me alone"? 

I have multiple dimmers, so my log's are spammed with 95% OTA messages.. Really annoying while trying to debug something else.

### What did you expect to happen?

Tell device to stop OTA requesting.

### How to reproduce it (minimal and precise)

Pair and wait.

### Zigbee2MQTT version

1.30.0-1

### Adapter firmware version

0x26680700

### Adapter

Conbee II

closed 2023-03-13 by github-actions[bot]

chojnicki, 2023-03-13:
@Koenkk can be reopened?

chojnicki, 2023-03-13:
@Koenkk I know, but like I said - it is not doing such thing :(

Just request, and no response at all.

Debug 2023-03-13 18:36:12Device 'living_room_cabinet_light' requested OTA

Koenkk, 2023-03-14:
Try to reduce the interval by adding the following to your `configuration.yaml`.

```yaml
ota:
  update_check_interval: 0
```

chojnicki, 2023-03-15:
@Koenkk tested on normal and edge version:

GL-C-009P
firmware build date 20220727
firmware version 10651203
 
> Debug 2023-03-15 16:35:42Received Zigbee message from 'living_room_ceiling_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":369176577,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID null
> Info 2023-03-15 16:35:42MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:35:42.206Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:35:42Device 'living_room_ceiling_light' requested OTA
> Debug 2023-03-15 16:35:42Failed to check if update available for 'living_room_ceiling_light' (No image available for imageType '5142')
> Info 2023-03-15 16:35:42MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:35:42.206Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:12Received Zigbee message from 'living_room_ceiling_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":369176577,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID null
> Info 2023-03-15 16:36:12MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:36:12.202Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:12Device 'living_room_ceiling_light' requested OTA
> Debug 2023-03-15 16:36:12Failed to check if update available for 'living_room_ceiling_light' (No image available for imageType '5142')
> Info 2023-03-15 16:36:12MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:36:12.202Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:42Received Zigbee message from 'living_room_ceiling_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":369176577,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID null
> Info 2023-03-15 16:36:42MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:36:42.210Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:42Device 'living_room_ceiling_light' requested OTA
> Debug 2023-03-15 16:36:42Failed to check if update available for 'living_room_ceiling_light' (No image available for imageType '5142')
> Info 2023-03-15 16:36:42MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_light', payload '{"brightness":129,"last_seen":"2023-03-15T15:36:42.210Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'

GL-C-009P (but really it is 5in1 dimmer that changes model name while changing mode).
firmware build date 20220418
firmware version 10551203

> Debug 2023-03-15 16:35:31Received Zigbee message from 'entrance_ceiling_lamp', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":352399361,"imageType":5145,"manufacturerCode":4687}' from endpoint 11 with groupID null
> Info 2023-03-15 16:35:31MQTT publish: topic 'zigbee2mqtt/entrance_ceiling_lamp', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-15T15:35:31.510Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":352399361,"latest_version":352399361,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:35:31Device 'entrance_ceiling_lamp' requested OTA
> Info 2023-03-15 16:35:31MQTT publish: topic 'zigbee2mqtt/entrance_ceiling_lamp', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-15T15:35:31.510Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":352399361,"latest_version":352399361,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:01Received Zigbee message from 'entrance_ceiling_lamp', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":352399361,"imageType":5145,"manufacturerCode":4687}' from endpoint 11 with groupID null
> Info 2023-03-15 16:36:01MQTT publish: topic 'zigbee2mqtt/entrance_ceiling_lamp', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-15T15:36:01.511Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":352399361,"latest_version":352399361,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:01Device 'entrance_ceiling_lamp' requested OTA
> Info 2023-03-15 16:36:01MQTT publish: topic 'zigbee2mqtt/entrance_ceiling_lamp', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-15T15:36:01.511Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installe [... cut ...] version":352399361,"latest_version":352399361,"state":"idle"},"update_available":false}'
> Debug 2023-03-15 16:36:31Device 'entrance_ceiling_lamp' requested OTA
> Info 2023-03-15 16:36:31MQTT publish: topic 'zigbee2mqtt/entrance_ceiling_lamp', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-15T15:36:31.510Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":352399361,"latest_version":352399361,"state":"idle"},"update_available":false}'

Koenkk, 2023-03-15:
Could you provide the herdsman debug logging starting from the point where you start z2m until the error + 2 minutes after that?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

chojnicki, 2023-03-16:
@Koenkk I'm not sure how to do that on HA, if it's even possible. Will try to access docker container. 

But next day I noticed something new, even that I did not changed anything. Now there is `Responded to OTA` in logs, but devices keep asking for OTA anyway, exactly every minute. I have 5 dimmers (planning to add more), so 5*60*24 = 7200 requests per day and clogged logs :/ Dimmers are pretty new.

Koenkk, 2023-03-18:
I see, then this cannot be fixed from the z2m side (should be fixed in the device firmware)

DenDeXTeR51, 2023-05-02:
Have same problem with OTA spam :(

catduckgnaf, 2023-09-12:
Same issue, my question is, can we filter and somehow disable the entity from requesting at all?

Gerben321, 2024-01-08:
I just updated my Z2M and now my 2 devices tell me there's a firmware update. However when I try to update I get this: Images currently unavailable.

vinnie99999, 2024-04-17:
I had almost 100 lights requesting for a OTA update (every 1 min and some every 30 sec) what was causing my coordinator to crash almost every day. After mailing with Gledopto support, they provided me with a firmware that only requested an update every 60min (still quite frequently but acceptable) and since than my Zigbee network is stable. But your light or controller needs to support OTA updates. I have still some lights laying around which i can't update :(

catduckgnaf, 2024-04-17:
Can you please share a link to that firmware? I emailed them a bit back and never got anything

vinnie99999, 2024-04-17:
I don’t have the firmware for your models. What models do you have?
I am using some spots GL-S-006P and some other models.
It took me more than 6 months and several email, before they started to reply. The 2nd and the 3th time I needed them it also took me months before I finally got what I needed.

I would say keep spamming them! I believe I finally got an answer when I started to email to this email address ***@***.***

"Organization", "name": "GitHub", "url": "https://github.com" } } ]

catduckgnaf, 2024-04-17:
I have GL-C-006P, didn't see what email you used

vinnie99999, 2024-04-17:
[image]
sales@gledopto.com is what i used in the past.

catduckgnaf, 2024-04-17:
thank you, i will try again

vinnie99999, 2024-04-17:
> thank you, i will try again

You're welcome. I don't know how to send a pm, but can you remove the quote reply text in your previous message? I see a lot of info i don't want to have there because i replied by email :( Thank you

catduckgnaf, 2024-04-17:
> > thank you, i will try again
> 
> You're welcome. I don't know how to send a pm, but can you remove the quote reply text in your previous message? I see a lot of info i don't want to have there because i replied by email :( Thank you

updated., thanks again, i tried emailing again

DenDeXTeR51, 2024-04-18:
[GL-S-004P](https://www.zigbee2mqtt.io/devices/GL-S-004P.html#gledopto-gl-s-004p) is spamming me for years, now I will spam their Support too for firmware:)

cdebavel, 2024-04-20:
Reddit thread on the topic: [https://www.reddit.com/r/Gledopto/comments/1c5oz6w/gledopto_lights_and_controllers_spamming/](https://www.reddit.com/r/Gledopto/comments/1c5oz6w/gledopto_lights_and_controllers_spamming/)

I have a GL-C-008-1ID controller. This is the email response I got:

>GL-C-008-1ID is not pro version, It cannot be upgraded,We suggest that you use the Pro version controller or ignore this information.

So I guess just have to live with it. Pretty ridiculous that something that _cannot_ update is requesting updates.

DenDeXTeR51, 2024-04-20:
I receive firmware for [GL-S-004P](https://www.zigbee2mqtt.io/devices/GL-S-004P.html#gledopto-gl-s-004p) with OTA requested changed to 1 hour...

JonPusey, 2024-08-12:
I received an updated firmware but am having trouble updating it, I have created the index file and added it to configuration.yaml, is there any missing step?

vinnie99999, 2024-08-13:
> I received an updated firmware but am having trouble updating it, I have created the index file and added it to configuration.yaml, is there any missing step?

This my ota: config in the configuration.yaml for z2m 
ota:
  ikea_ota_use_test_url: false
  disable_automatic_update_check: false
  zigbee_ota_override_index_location: my_index.json
  update_check_interval: 1440

Create a file named my_index.json under the zigbee2mqtt folder. For me the config looks like this (layout is a bit scrambled because of the post, but make sure it is in yaml layout when you save the file) also make sure the parameters correspond to your ota file, if i remember it right there should be a script / command line tool which can help you with it (is documented somewhere :) ):
[
    {
        "fileVersion":604057601,
        "fileSize":209330,
        "manufacturerCode":4687,
        "imageType":5172,
        "sha512":"672fb20c0cfc9650f0a13d0fbeb50367d5239b090eb0bc9da5b1a7ec69616f7bb080e0ff6af43824bb7cfac0a6ed0bddcd0b838a115945546a0cdfa11a83627f",
        "url":"ota/GL-S-006P_V20451203_20240124_ota60min.ota",
        "force": false
    },
    {
        "fileVersion":604057601,
        "fileSize":209330,
        "manufacturerCode":4687,
        "imageType":5170,
        "sha512":"3b4d0acb78c8d282080f8036b73efd86668fdacfe6bb97f14e6601d87096c1db16f0a2a2653104a697faf8ce76618038a9e70fb266ca25c785769b7e3c76f258",
        "url":"ota/GL-S-004P_V20451203_20240124_ota60min.ota",
        "force": false
    }
]

Under the zigbee2mqtt folder i have created a ota folder and copied the ota file to it.

If i remember it correctly i used the deconz update procedure from Gledopto first. Newer test OTA updates i did with z2m because i got tired of unplugging and repairing my lights.

I hope this helps

KHV8, 2024-11-01:
> I received firmware for [GL-S-004P](https://www.zigbee2mqtt.io/devices/GL-S-004P.html#gledopto-gl-s-004p) with OTA requested changed to 1 hour...

[DenDeXTeR51](https://github.com/DenDeXTeR51), any possibility you can share the firmware. I have the same spot, and the  same problem, they update every 30sec.