Koenkk/zigbee2mqtt #16514: [Feature request]: Add invert open/close commands on cover for Smartwings WM25L-Z

latez, 2023-02-04:
### Is your feature request related to a problem? Please describe

I will state outright, that I am copying Request #13982 as my request is identical.

These SmartWings Home roller shades are becoming extremely popular, and while the device is useable in its - the commands are reversed (open actually closes it and close actually opens it)

### Describe the solution you'd like

Ideally the shades would function as they were designed and directional functionality restored as it is incredibly difficult to work with in its current state. 

### Describe alternatives you've considered

I've tried reversing the cover with the remote but this causes other issues with the states and the remote itself. (Yes I did this as well, and it sucks)

### Additional context

Appreciate your consideration to resolving truly crippling issue! thanks

latez, 2023-03-12:
Still hoping someone looks at this at some point - thank you!

mikerenna, 2023-03-12:
Just got a set of the smartwings as a test before buying them for the whole house. Disappointed to discover that the open/close state and buttons are inverted. If i'm reading this item correctly, this is the same request. There is an invert_cover setting that appears to Invert the cover position, false: open=100,close=0, true: open=0,close=100. Can this same flag be used to flip the direction (and state value) of the open/close buttons ?  I would agree that this brand will do well. i personally was looking for something this good out of box for years.

latez, 2023-03-12:
Testing with this did not yield positive results unfortunately. You can reverse the shades (they physically flip) and the controls become reversed on the physical remote which unfortunately makes that useless. Ideally you would be able to invert strictly within Z2M - as it is I've been forcing the family to remember open means close and vice versa 🥲

mikerenna, 2023-03-12:
That's the situation I'm in now. Any automation with home assistant, i believe I will have to call services for cover.open to close it and vice versa. Also, presumably any voice control will painful as you'd be using the opposite commands. 100% agree that an option to reverse it in Z2M would be a life saver here.

mikerenna, 2023-03-14:
Maybe we need to find out koenkk's window dimensions and send him a set as incentive. :)

mikerenna, 2023-04-07:
Finally found an answer. This fixed it for me!!
[image]

latez, 2023-04-07:
Glad you were able to get that resolved! Unfortunately that hasn't done anything for me - I've done it several times and the motor jogs as its supposed to but it's not clear that it has any affect. Up is down and down is up :)

mikerenna, 2023-04-11:
make sure you don't miss the second step of taking off the back cover and hitting the program button. i was missing that the first time around and just doing the double button press which flipped both the remote and zigbee. It wasn't until i did the second step here that it brought them into sync.

closed 2023-05-19 by github-actions[bot]

robweidner, 2023-06-24:
@mikerenna I noticed that doing the motor direction swap using the remote, swaps the directions on the physical remote as well.  My wife doesn't like that :) are you able to swap the direction of the motor and then swap the direction on the remote so it works for both Home Assistant and the remote?

mikerenna, 2023-06-24:
When I talked to support they gave me a sequence that swapped only the zigbee one and it worked perfectly. Completely fixed the issues. I think it's in the manual but was easy to misread. Let me see if I can find it.

robweidner, 2023-06-24:
That would be amazing. It seems as though when I do the reverse
swap using the remote it swaps the remote direction as well.
Therefore it would be amazing to only swap in app :) let me know
if you figure it out!

Best regards,
--
Rob Weidner
Automation & Systems Integrator
+1.224.634.5814
--
Curious about my current tech stack?
Missive, Chat GPT-4, VimCal, Discord, Slack, Notion, Airtable,
Stacker, Make, Xero, Cognito Forms, Shopify, Octoparse,
ParseHub, and Grammarly are some of my favorite tools I use
Ask me about my current setup at home!
Home Assistant, Sonos, Roku, TP-Link Kasa, Amazon Fire, Wyze,
Apple, Google, Ubiquiti

mikerenna, 2023-06-24:
i just looked and I think my april 7 response was the correct sequence. i was having the same problem as you and whatever sequence I was doing, was changing both, but then SmartWings support walked me through it and i realized i was missing a step. I don't recall exactly what, but try it again and make sure you do everything you see on the page above. If that doesn't work, reach out to smartwings support, They were very helpful (the support person reached out to their head engineer who totally understood the issue and advised me well). Mine works perfectly now so I can attest that there is a simple fix.

rmdrew88, 2023-09-24:
Chiming in here - I just switched from SmartThings to Home Assistant, and added my Smartwings roller blinds via Zigbee2MQTT.  @mikerenna - your directions fixed the revered state issue for me perfectly.  As an added bonus, the remote directions are still correct as well.  The key is to press the up and down buttons FIRST, then press the "p" button behind the cover SECOND.  Other instructions I have found only reverse those directions.

highincolorado, 2024-04-12:
> Finally found an answer. This fixed it for me!! [image]

Currently in a discussion with SmartWings support and they confirm this solution but tell me they made a hardware change in 2023 that this fix addresses and older units do not have this function.