Koenkk/zigbee2mqtt #16515: [Feature request]: Add Avatto ME167 manufacturer

vingerha, 2023-02-04:
### Is your feature request related to a problem? Please describe

Bought this:
https://fr.aliexpress.com/item/1005003952150617.html?spm=a2g0o.order_list.order_list_main.29.3cd85e5by0tuuO&gatewayAdapt=glo2fra

Manufacturer: _TZE200_6rdj8dzm
I managed the workaround via extension me167.js but it would be nice to have it in the main package too

### Describe the solution you'd like

Recognise manufacturer: _TZE200_6rdj8dzm

### Describe alternatives you've considered

extension

### Additional context

I managed the workaround via extension me167.js but it would be nice to have it in the main package too

Koenkk, 2023-02-04:
`_TZE200_6rdj8dzm` is supported in z2m 1.30.0

closed 2023-02-04 by Koenkk

morzini, 2024-11-19:
Hi, is it possible to also add _manufacturer: TZE284_o3x45p96. I brought here https://fr.aliexpress.com/item/1005006439482487.html?spm=a2g0o.order_list.order_list_main.5.36535e5b8nvuHh&gatewayAdapt=glo2fra

Thanks a lot.

Morzini