Koenkk/zigbee2mqtt #16516: [Feature request]: App Package to yunohost

Qestions, 2023-02-04:
### Is your feature request related to a problem? Please describe

I`m loocking for an simple way to use a Smarthome and other services on the same Hardware in my local Network

### Describe the solution you'd like

Cloud you pleas offer an App-Package to yunohost? 
https://yunohost.org/
It will make the way to use Zigbee2mqtt easier for a lot of users. 
There are already users who wish to use zigbee2mqtt there:
https://yunohost.org/de/apps_wishlist
https://yunohost.org/de/packaging_apps_intro

### Describe alternatives you've considered

The only way I currently see to use Zigbee2mqtt is, to get it by pulling it strait from git.
This is no secure way, when I`m using yunohost in public.

### Additional context

-

junosuarez, 2023-03-07:
You mention you're currently running by pulling the source from github.

An easier way for self-hosting is using the official docker image, see https://www.zigbee2mqtt.io/guide/installation/02_docker.html

closed 2023-04-14 by github-actions[bot]

jedie, 2024-01-14:
I started to implement this here: https://github.com/YunoHost-Apps/zigbee2mqtt_ynh

Looks promising!

Discuss also here: https://forum.yunohost.org/t/yunohost-package-for-zigbee2mqtt/28039