Koenkk/zigbee2mqtt #16520: TS110E dimmer does not expose min_brightness and max_brightness

mikolajpiecuch, 2023-02-04:
### What happened?

TS110E dimmer (_TZ3210_k1msuvg6) does not expose min_brighntess and max_brightness in frontend even state of that dimmer shows that fields:

`{
    "brightness": 253,
    "linkquality": 39,
    **"max_brightness": 255,
    "min_brightness": 4,**
    "power_on_behavior": "previous",
    "state": "OFF",
    "switch_type": "momentary"
}`

### What did you expect to happen?

Possibility to set min_brightness for TS110E dimmer - physical switch won't be able to reduce brightness below set threshold

### How to reproduce it (minimal and precise)

Add TS110E dimmer to zigbee2mqtt

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20211115

### Adapter

CC2531

Koenkk, 2023-02-04:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-04 by Koenkk

jacekmarek, 2023-02-06:
@mikolajpiecuch I have this module _TZ3210_k1msuvg6 also but I can't use brighntess  as trigger in automation Can you check this?

mikolajpiecuch, 2024-11-16:
I found this commit removed switch type for _TZ3210_k1msuvg6 -> https://github.com/Koenkk/zigbee-herdsman-converters/commit/401d1c4be1690c89c9d05dd69ce998af284484bc. I'm not sure if can add only this line
`tuya.exposes.switchType()`
as it can affect other devices. I guess default value for the switch type is 'momentary' and that's why it works for most users. I still use 'toggle' switch so I need to have this option set