Koenkk/zigbee2mqtt #16521: 1.30.0 breaks the device "Datek 0402946". It gets listed as unsupported after update

Megamannen, 2023-02-04:
### What happened?

I updated to 1.30.0, my Zigbee smart lock stopped working (listed as unsupported)

### What did you expect to happen?

That it still is supported

### How to reproduce it (minimal and precise)

Have  Datek 0402946 and run 1.29, then upgrade to 1.30.
Or have 1.30 and see that it's unsupported, then downgrade to 1.29 and see how it becomes supported once again

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20210708

### Adapter

Sonoff

Megamannen, 2023-02-04:
Might be this fix I guess?

* Updated Datek sensors & new converter to read HW version: https://github.com/Koenkk/zigbee-herdsman-converters/pull/5310

Koenkk, 2023-02-04:
This will be fixed in tomorrows hotfix release.

closed 2023-02-04 by Koenkk