Koenkk/zigbee2mqtt #16529: Saswell SEA801-Zigbee/SEA802-Zigbee - child lock function not working

MihaiKrieger, 2023-02-04:
### What happened?

I have [this ](https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html) device. 
The `child_lock` function does not work. The toggle works fine and it looks like there is no reason why it would not work.
However, the TRV values can be physically changed by rotating the tip. 

`child_lock` being locked or unlocked makes no difference.

### What did you expect to happen?

Expecting the TRV to have the physical interaction disabled and only be controllable via Z2M / remotely.

### How to reproduce it (minimal and precise)

1) Activate Child Lock
[image]

2) Physically turn the tip of the TRV. Expecting the values to **NOT** change. However, they still change.

### Zigbee2MQTT version

 1.30.0

### Adapter firmware version

20220219

### Adapter

 Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

`Info 2023-02-04 21:08:56Succesfully changed options
Info 2023-02-04 21:08:56MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"yv1w6-1"}'
Debug 2023-02-04 21:09:04Received MQTT message on 'zigbee2mqtt/Baby Room Radiator/set/child_lock' with data 'UNLOCK'
Debug 2023-02-04 21:09:04Publishing 'set' 'child_lock' to 'Baby Room Radiator'
Debug 2023-02-04 21:09:09Received Zigbee message from 'Baby Room Radiator', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":40}],"seq":31}' from endpoint 1 with groupID 0
Info 2023-02-04 21:09:09MQTT publish: topic 'zigbee2mqtt/Baby Room Radiator', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":25,"frost_detection":"ON","heating":null,"last_seen":"2023-02-04T21:09:10+02:00","linkquality":36,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Debug 2023-02-04 21:09:10Received Zigbee message from 'Balcony Desk Lamp', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 1 with groupID 0
Info 2023-02-04 21:09:10MQTT publish: topic 'zigbee2mqtt/Balcony Desk Lamp', payload '{"brightness":129,"color_mode":"color_temp","color_temp":289,"last_seen":"2023-02-04T21:09:11+02:00","linkquality":43,"power_on_behavior":null,"state":"OFF","update":{"installed_version":69635,"latest_version":69635,"state":"idle"},"update_available":null}'
Debug 2023-02-04 21:09:10Succesfully pinged 'Balcony Desk Lamp' (attempt 1/2)
Debug 2023-02-04 21:09:24Received MQTT message on 'zigbee2mqtt/Baby Room Radiator/set/child_lock' with data 'LOCK'
Debug 2023-02-04 21:09:24Publishing 'set' 'child_lock' to 'Baby Room Radiator'
Debug 2023-02-04 21:09:25Received Zigbee message from 'Baby Room Radiator', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":103}],"seq":768}' from endpoint 1 with groupID 0
Info 2023-02-04 21:09:25MQTT publish: topic 'zigbee2mqtt/Baby Room Radiator', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":25,"frost_detection":"ON","heating":null,"last_seen":"2023-02-04T21:09:26+02:00","linkquality":32,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Debug 2023-02-04 21:09:25Received Zigbee message from 'Baby Room Radiator', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":40}],"seq":32}' from endpoint 1 with groupID 0
Info 2023-02-04 21:09:25MQTT publish: topic 'zigbee2mqtt/Baby Room Radiator', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCK","current_heating_setpoint":25,"frost_detection":"ON","heating":null,"last_seen":"2023-02-04T21:09:26+02:00","linkquality":32,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Debug 2023-02-04 21:09:30Received Zigbee message from 'Baby Room Radiator', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":103}],"seq":768}' from endpoint 1 with groupID 0
Info 2023-02-04 21:09:30MQTT publish: topic 'zigbee2mqtt/Baby Room Radiator', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCK","current_heating_setpoint":25,"frost_detection":"ON","heating":null,"last_seen":"2023-02-04T21:09:31+02:00","linkquality":32,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Debug 2023-02-04 21:09:43Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"yv1w6-2"}'
Info 2023-02-04 21:09:44Succesfully changed options
Info 2023-02-04 21:09:44MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"yv1w6-2"}'
Info 2023-02-04 21:10:09MQTT publish: topic 'zigbee2mqtt/Desk Lamp', payload '{"brightness":102,"color_mode":"color_temp","color_temp":445,"last_seen":"2023-02-04T21:10:10+02:00","linkquality":105,"power_on_behavior":null,"state":"ON","update":{"installed_version":69635,"latest_version":69635,"state":"idle"},"update_available":null}'
Info 2023-02-04 21:10:30MQTT publish: topic 'zigbee2mqtt/Kitchen Window Sensor', payload '{"battery":97,"contact":true,"device_temperature":22,"last_seen":"2023-02-04T21:10:31+02:00","linkquality":116,"power_outage_count":4,"voltage":2995}'`

closed 2023-03-14 by github-actions[bot]

shitrip, 2023-10-02:
Confirmed. This used to work, although I do recall there being an issue toggling the parameter. I'm unable to make the switch work no matter what nowadays.

shitrip, 2023-10-02:
related issue https://github.com/Koenkk/zigbee2mqtt/issues/9375