Koenkk/zigbee2mqtt #16535: power_on_behavior exposed although unsupported by LIDL plug

N3rdix, 2023-02-05:
### What happened?

After upgrading to 1.30.0 my LIDL smart plugs (https://www.zigbee2mqtt.io/devices/HG06337.html#lidl-hg06337) expose the attribute power_on_behavior which seems to not be supported by that adapter. When using Home Assistant that leads to unavailable entities for that attribute.

Can this be fixed somehow on Z2M side?

### What did you expect to happen?

The attribute should not be exposed for adapters which do not support it

### How to reproduce it (minimal and precise)

Refresh attribute power_on_behavior or try to set a value for it

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

```
Debug 2023-02-05 09:39:13Received MQTT message on 'zigbee2mqtt/Smart Plug Drucker/set' with data '{"power_on_behavior":"previous"}'
Debug 2023-02-05 09:39:13Publishing 'set' 'power_on_behavior' to 'Smart Plug Drucker'
Error 2023-02-05 09:39:13Publish 'set' 'power_on_behavior' to 'Smart Plug Drucker' failed: 'Error: Write 0x5c0272fffe2c5481/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2023-02-05 09:39:13Error: Write 0x5c0272fffe2c5481/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:111:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
```

Koenkk, 2023-02-05:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {manufacturerName: '_TZ3000_kdi2o9m6'}, // EU
        {modelID: 'TS011F', manufacturerName: '_TZ3000_plyvnuf5'}, // CH
        {modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}, // FR
        {modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}, // BS
        {modelID: 'TS011F', manufacturerName: '_TZ3000_upjrsxh1'}, // DK
        {manufacturerName: '_TZ3000_00mk2xzy'}, // BS
    ],
    model: 'HG06337',
    vendor: 'Lidl',
    description: 'Silvercrest smart plug (EU, CH, FR, BS, DK)',
    extend: tuya.extend.switch(),
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(11);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

N3rdix, 2023-02-05:
Thanks Koen, you do an amazing job (time for a donation now 🥇)!

The converter fixes the issue and the attribute can be read/changed.

In case it is helpful this is the debug log when i refreshed the state (which seems to default to "previous"):
```
Debug 2023-02-05 15:56:55Received MQTT message on 'zigbee2mqtt/Smart Plug Drucker/get' with data '{"power_on_behavior":""}'
Debug 2023-02-05 15:56:55Publishing get 'get' 'power_on_behavior' to 'Smart Plug Drucker'
Debug 2023-02-05 15:56:56Received Zigbee message from 'Smart Plug Drucker', type 'readResponse', cluster 'genOnOff', data '{"moesStartUpOnOff":2}' from endpoint 1 with groupID 0
Info 2023-02-05 15:56:56MQTT publish: topic 'zigbee2mqtt/Smart Plug Drucker', payload '{"linkquality":128,"power_on_behavior":"previous","state":"OFF"}'
```

Koenkk, 2023-02-06:
Thanks, integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)