Koenkk/zigbee2mqtt #16539: Not executing commands when connected to coordinator

TheLion, 2023-02-05:
### What happened?

One of my Danfoss Radiator Valves (014G2461) does not execute commands send from Z2M when it is connected to my coordinator (Electrolama zig-a-zig-ah!). It will produce an error in Z2M with a timeout. It does send updates though, like when the setpoint is manually changed on the radiator valve.

When I force it to connect to one of the routers or power Zigbee devices, (by unplugging the coordinator for a couple of seconds), it works just fine.
Until if finds it way back to the coordinator, as it is the closest device as they are in the same room.

I tried repositioning the Coordinator, both closer to and further away from the Danfoss device, but that doesn't make a difference.

The strange thing is that is does send updates, but does not process updates send from Z2M.

Other devices connected to the coordinator, don't have this issue, not even when it is another Danfoss radiator valve.

I'm running the latest firmware on my Coordinator and routers and Z2M is the latest version, 1.30.0, but this issue didn't start with the version, it happened before.

Any thoughts on how to troubleshoot/solve this?

### What did you expect to happen?

I expected that it wouldn't make a difference to which device/coordinator/router a device is connected. I should behave the same.

### How to reproduce it (minimal and precise)

Unplug the device this specific Danfoss device is connected to, and it will likely connect to the coordinator. If not, keep repeating till it does.

Then send a command from Z2M (or Domoticz for that matter) and see what happens. It will produce an error with a timeout.

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah!

lux73, 2023-02-05:
> I'm running the latest firmware on my Coordinator and routers and Z2M is the latest version, 1.30.0, but this issue didn't start with the version, it happened before.
> 
not anymore -> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin there is an new Firmware from 20221226
> Any thoughts on how to troubleshoot/solve this?
> ### What did you expect to happen?
> 
i had the same expierence with one of my newly purchased Danfoss Ally - 3 of 4 works without hickups, but one of them had the same Problem as you described

after flashing the new Firmware on my CC2652RB & newly re-pairing this Valve the issue is gone :-)

TheLion, 2023-02-05:
> not anymore -> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin there is an new Firmware from 20221226

Oh, missed that.

> i had the same expierence with one of my newly purchased Danfoss Ally - 3 of 4 works without hickups, but one of them had the same Problem as you described
>
> after flashing the new Firmware on my [...]

Thanks!. This would be an easy fix. I followed your steps and will monitor its behavior.

hwinkel, 2023-02-07:
HI, have you tested this with the Athom Zigbee Bridge as well? 

https://es.athom.tech/blank-1/zigbee-gateway?utm_campaign=41d0f227-1be7-4ca8-b4c3-64fa6754fbb8&utm_source=so&utm_medium=mail&cid=5ef89e61-e3a9-4a0c-8e75-87d35d3bc8f6

And how the Firmware gets updated on this Ethernet Bridge Gateway? is Z2M updating the firmware on such gateways? or is this a local action?

Context: I have received a Danfoss Thermostat as well, and it's loosing connection after a few hours with Z2M (ZHA has worked longer) Also, seems Danfoss has a new 0.20 FW Version, at least on my device:

[image]

TheLion, 2023-02-07:
> HI, have you tested this with the Athom Zigbee Bridge as well?

No, I haven't. I don't currently have that bridge.

> Context: I have received a Danfoss Thermostat as well, and it's loosing connection after a few hours with Z2M (ZHA has worked longer) Also, seems Danfoss has a new 0.20 FW Version, at least on my device:

The Zigbee model of my Danfoss is different, eTRV010 and there is no update available for this version.

lux73, 2023-02-07:
@hwinkel 

your Danfoss is not completely paired as you can see on your Screenshot right bottom:

`Interview completed     False`

this should be `True`

two of my newly purchased Ally's have Firmware 00.20 too - have found anywhere that's related to new/other Hardware in the new Ally (no added Functions or anything new in this 00.20 Firmware)

lux73, 2023-02-07:
>  I even made sure that the Ally is directly connected to the Coordinator, by switching ALL router device off while pairing.
> 
don't do this - Zigbee Devices designed to search and reconnect to the nearest Router. At least the Danfoss do this without Problems (other than Aqara/Xiaomi Devices)

so turn on all Router and re-pair your Danfoss at the Location it will be

> But anyway, thanks for the feedback. I'll try further to get the Interview done, if you say you managed it. May I should change to another coordinator stick, as I still don't know how the FW updates are managed with a TCP connected Stick.
>
don't know your mentioned Network Connected Athom Coordinator. But i would suspect it's not really stable over time.
i have made some suspicious experience with ser2net - could never got it running stable (it's already the same with your USB to TCP connected Coordinator)

i would suggest to buy a CC2652xx Coordinator directly connected to USB-A Port. **Be aware to buy a Sonoff Dongle-E, afaik the Firmware is experimental and also unstable**

i'm using Sleash's CC2652RB with Firmware 20221226 - works without hickups

TheLion, 2023-02-19:
After the upgrade of the firmware on my coordinator, all seems to work just fine.

closed 2023-02-19 by TheLion

hwinkel, 2023-02-19:
> After the upgrade of the firmware on my coordinator, all seems to work just fine. 
> 
> 

How do you updated the firmware on the athom bridge ?

TheLion, 2023-02-19:
> How do you updated the firmware on the athom bridge ?

I didn't. My issue was not with a Athom bridge.