Koenkk/zigbee2mqtt #16543: Poor scene transition time with Ikea TRADFRI driver

Kevguy91, 2023-02-05:
### What happened?

When creating a group scene with a specific transition duration, the transition duration is not respected by the device (https://www.zigbee2mqtt.io/devices/ICPSHC24-10EU-IL-1.html#ikea-icpshc24-10eu-il-1). No problem for the other bulbs in the group (Phillips Hue)

### What did you expect to happen?

A transition duration as defined during creation.

### How to reproduce it (minimal and precise)

I created the group scene with the transition following the instructions on this page : https://www.zigbee2mqtt.io/guide/usage/scenes.html#using-transitions-in-scenes

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle - CC2652P

### Debug log

Debug 2023-02-05 15:53:24Received MQTT message on 'zigbee2mqtt/Salle à manger / Lumière salle à manger/set' with data '{"scene_recall":0}'
Debug 2023-02-05 15:53:24Publishing 'set' 'scene_recall' to 'Salle à manger / Lumière salle à manger'
Info 2023-02-05 15:53:25MQTT publish: topic 'zigbee2mqtt/Salle à manger / Lumière buffet', payload '{"brightness":254,"color":{"h":43,"hue":43,"s":36,"saturation":36,"x":0.3608,"y":0.3635},"color_mode":"color_temp","color_temp":222,"last_seen":"2023-02-05T15:53:17+01:00","level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":"previous","state":"ON","update":{"installed_version":587753009,"latest_version":587753009,"state":"idle"},"update_available":null}'
Info 2023-02-05 15:53:25MQTT publish: topic 'zigbee2mqtt/Salle à manger / Lumière plafonnier', payload '{"brightness":239,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"color_temp_startup":370,"last_seen":"2023-02-05T15:53:18+01:00","linkquality":94,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16784918,"latest_version":16784918,"state":"idle"},"update_available":null}'
Info 2023-02-05 15:53:25MQTT publish: topic 'zigbee2mqtt/Salle à manger / Lumière salle à manger', payload '{"brightness":254,"color":{"h":43,"hue":43,"s":36,"saturation":36,"x":0.3608,"y":0.3635},"color_mode":"color_temp","color_temp":222,"state":"ON"}'
Info 2023-02-05 15:53:25MQTT publish: topic 'zigbee2mqtt/Salle à manger / Lumière salle à manger', payload '{"brightness":239,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"state":"ON"}'

Kevguy91, 2023-03-10:
Nobody has this problem?

Kevguy91, 2023-03-21:
I did some tests :

Device transition setting : no problem
Group transition setting : no problem
Device scene recall : transition problem
Group scene recall : transition problem

I saw that a coordinator update was available, can this change a few things?

Kevguy91, 2023-04-20:
I revive the topic, does anyone have this problem?

closed 2023-05-29 by github-actions[bot]

MoLiNaSa, 2024-01-22:
> I revive the topic, does anyone have this problem?

same issue here, a recall ignores any kind of transition settings

giga8yte, 2024-01-27:
> I revive the topic, does anyone have this problem?

Ive got the same issue. Mine are Sengled. Weird thing is in the same group of 9 bulbs 2 of those are at the proper transition and the rest are no transition at all.  ver 1.35.1-1