Koenkk/zigbee2mqtt #16546: LQI low with Sonoff Zigbee Dongle Plus

Drealine, 2023-02-05:
### What happened?

Hi, 

Since migrated from Conbee II to Sonoff Zigbee Dongle Plus (for timeout reason of lighs philipe hue), my LQI drastic change from 250 to 120 max.
I don't have performance problem since 2 days but I would like to know if there is a problem or just a problem to calculating LQI ?
Of course, I flashed my USB stick to the latest Zstack firmware and set transmit power to 20 in frontend.

Thank you :) 

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20221226

### Adapter

Sonoff zigbee dongle plus CC2652P

Koenkk, 2023-02-06:
The Conbee LQI calculation is not correct. A value of 120 is perfectly fine.

Drealine, 2023-02-06:
That's great! 
Maybe good to write a comment in the adapters page ? Because a lot of people which came from Conbee are losted about this parameter. 

Have a nice day :)

closed 2023-02-06 by Drealine