Koenkk/zigbee2mqtt #16547: Zigbee2MQTT on Linux

amgstone, 2023-02-05:
### What happened?

[image]

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210319

### Adapter

CC2538

dragondagda, 2023-02-06:
@amgstone can you add some logs, i.e. by executing `journalctl -u zigbee2mqtt.service` ?

ristoavila, 2023-02-08:
status=200/CHDIR means that systemd can't access the directory which you have specified in the service configuration. Do you have typo there or is it missing or wrong access rights?

amgstone, 2023-02-08:
already done everything, everything is fine. what command to update zigbee2mqtt?

ristoavila, 2023-02-08:
Depends how you installed zigbee2mqtt. If you installed it directly using git please see https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version otherwise refer to other guides.

amgstone, 2023-02-13:
how to make automatic start?

closed 2023-03-23 by github-actions[bot]