Koenkk/zigbee2mqtt #16553: Missing effect support from some of the Philips devices e.g. 1743030P7 and 1742930P7

ristoavila, 2023-02-06:
### What happened?

Philips has added the candle light and other effects to their hue series and some models are missing these.
E.g. I'm using 1743030P7 and 1742930P7 lamps and those miss new effects which are part of the firmware updates.

I also found a list describing which models supports effects
https://www.assets.signify.com/is/content/Signify/Assets/hue/united-states/20220124-effects-compatible-lights.pdf

### What did you expect to happen?

I would expect under device effects there are newly added effects listed.

### Zigbee2MQTT version

1.30

### Adapter firmware version

20221226

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

Koenkk, 2023-02-06:
@zegl @sjorge what would be wise here; adding the effects for every bulb is a lot of work, would it be an idea to enable it for all lights and then disable it again for those where it doesn't work for?

sjorge, 2023-02-06:
Is there a clear mapping between the ones listed in the pdf and how we list them?If so, I guess going over the while list by next month might be doable if we do in small batches.~ sjorge

ristoavila, 2023-02-06:
I would enable it to all philips lights and then disable it if some one complains it doesn't work. I would also expect philips to roll more effects in the futures and enable those through FW updates. 

There is also similar mapping as you have in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js e.g. one can find the model 1742930P7 from the previously posted web page with this exact model name.

I could also contribute to herdsman-converters but not sure how to enable effects as e.g. 1742930P7 is different from 1743030P7. First one has disableHueEffects and next one doesn't have any extra parameters.

ristoavila, 2023-02-06:
One more thing to mention is that I'm only checking the z2mqtt web UI and under there the effect drop down is missing the candle light etc in both cases 1742930P7 and 1743030P7.

sjorge, 2023-02-06:
So Art Code column seems to line up with the zigbeeModel. Let me take a crack at getting the ones from the PDF updated.

sjorge, 2023-02-06:
I'm about 3 pages in and it seems like so far it's like nearly all of them, turning it on by default and disabling as needed is indeed the way to go.

danieledwardgeorgehitchcock, 2023-02-06:
Hi all,

I believe some hue firmwares for white ambiance support candlelight and a couple of other effects but not the full range of effects that the colour bulbs support. Would the proposal and PR here work with both types of bulb?

ristoavila, 2023-02-07:
I still wonder where the gui gets the list of supported effects. The two models I have listed only few of the effects and all of the enum defined ones were missing e.g. [the changs the listing. e done removes the listing. ](https://github.com/Koenkk/zigbee-herdsman-converters/commit/660c88d15a7b8c5dab3176dccc819b9aea6859e6) this seem to remove all the items. I had everything else visible in the gui expect the candle and others defined in the enum here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/philips.js on line 45

sjorge, 2023-02-07:
> I still wonder where the gui gets the list of supported effects. The two models I have listed only few of the effects and all of the enum defined ones were missing e.g. [the changs the listing. e [...]

The few that used `light_onoff_brightness` I tested were not on the list in the PDF. The expose templates for `light_onoff_brightness_colortemp`, `light_onoff_brightness_color`, and `light_onoff_brightness_colortemp_color` have the list hardcoded, depending if there is just colortemp or colorxy support the set is different.

closed 2023-03-17 by github-actions[bot]