Koenkk/zigbee2mqtt #16554: Develco SMRZB-332 not working after update

Fatherjohn, 2023-02-06:
### What happened?

Upgraded to zStack3x0 1.30.0(CC2652P). Lost Develco SMRZB-332, Zigbee 2 mqtt can't find it. Rolled back to previous version and everything is ok

### What did you expect to happen?

That everything worked as before

### How to reproduce it (minimal and precise)

Flashed Sonoff Zigbee 3.0 USB Dongle Plus (model: ZBDongle-P) with latest firmware

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (model: ZBDongle-P)

Koenkk, 2023-02-06:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5425, hotfix release will be available today.

Fatherjohn, 2023-03-09:
Problem fixed, thanks.

closed 2023-03-09 by Fatherjohn