Koenkk/zigbee2mqtt #16555: Z2m not starting anymnore

JackTalisker, 2023-02-06:
### What happened?

My z2m instance does not start anymore

### What did you expect to happen?

works

### How to reproduce it (minimal and precise)

This morning I found my z2m instance offline.
This happened after a scheduled weekly reboot of my wifi router but I don't know if it is related to (last week after that reboot I lost all my ab devices and had to restore a previous backup).
I tried everything:
- recent backup restore
- recreating the docker image
- latest dev build

### Zigbee2MQTT version

1.30.0 dev

### Adapter firmware version

Tasmota 12..3.1.6 + ncp-uart-nsw_6.7.9_115200

### Adapter

Sonoff ZBBridge tasmota flashed

### Debug log

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-02-06 09:35:14: Logging to console and directory: '/var/log' filename: log.txt
Zigbee2MQTT:info  2023-02-06 09:35:14: Starting Zigbee2MQTT version 1.30.0-dev (commit #0e1220f)
Zigbee2MQTT:info  2023-02-06 09:35:14: Starting zigbee-herdsman (0.14.90)
Zigbee2MQTT:error 2023-02-06 09:35:35: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-06 09:35:35: Failed to start zigbee
Zigbee2MQTT:error 2023-02-06 09:35:35: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-06 09:35:35: Exiting...
Zigbee2MQTT:error 2023-02-06 09:35:35: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:524:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:326:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

JackTalisker, 2023-02-06:
My z2m configuration.yaml:

```
homeassistant:
  legacy_entity_attributes: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
  user: username
  password: password
serial:
  port: tcp://192.168.0.77:8888
  adapter: ezsp
advanced:
  log_directory: /var/log
  log_level: info
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601
  homeassistant_legacy_entity_attributes: false
  legacy_api: true
  legacy_availability_payload: true
experimental: {}
frontend:
  port: 8088
ota:
  update_check_interval: 10
  disable_automatic_update_check: false
devices:
[..........]
external_converters:
  - HS1RGB.js
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
device_options:
  legacy: false
```

My docker-compose.yaml:

```
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
    - "/home/pi/zigbee2mqtt:/app/data"
    devices:
    - /dev/null:/dev/ttyACM0
    environment:
    - TZ=Europe/Rome
   # - DEBUG=zigbee-herdsman*
    network_mode: host
    restart: always
```

JackTalisker, 2023-02-06:
I had to reissue this command to the tasmota console of my coordinator (after changed to "module 75"):

`backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0`

now z2m started again.
Don't know why happened this to the server mapping.

closed 2023-02-06 by JackTalisker

Dom4n, 2023-03-01:
@JackTalisker thank you very much! 

I had this same problem, and issuing this command to tasmota console solved it somehow.