Koenkk/zigbee2mqtt #16560: [New device support]:

loogfreak, 2023-02-06:
### Link

https://de.aliexpress.com/item/1005005122047276.html?spm=a2g0o.order_list.order_list_main.5.6de95c5f6p5u8Y&gatewayAdapt=glo2deu

### Database entry

   '0xa4c1385e8453a60d':     friendly_name: '0xa4c1385e8453a60d'

### Comments

Hi, I have zigbee2mqtt and home assistant running in docker. 
I have now ordered two sockets via Aliexpress and would like to integrate them into my Smarthome.
Unfortunately, the sockets are not supported in my zigbee2mqtt version. What can I do?
Modell: TS011F
Manufracturer: _TZ3000_hdopuwv6

Thanks for your help!

loogfreak, 2023-02-06:
My zigbee2mqtt version ist 1.26. 
I think i need 1.28.2. or newer, but after a docker update nothing happend

dragondagda, 2023-02-06:
@loogfreak How did you install zigbee2mqtt? With docker or docker compose?

Check the documentation for update instructions:
- Docker: https://www.zigbee2mqtt.io/guide/installation/02_docker.html#updating
- Docker compose: https://www.zigbee2mqtt.io/guide/installation/02_docker.html#updating-1

loogfreak, 2023-02-06:
> @loogfreak How did you install zigbee2mqtt? With docker or docker compose?
>
> Check the documentation for update instructions:
>
>     * Docker: [...]

Hey dragondagda, 

I have installed z2m as docker compose. 
When I am in the directory of the docker-compose.yml and want to do the update, I get the following message:
"services.zigbee2mqtt.ports must be a list".

What exactly does the message mean?

dragondagda, 2023-02-06:
It could be a formatting issue in your docker-compose.yml. Make sure you use only spaces and no tabs

loogfreak, 2023-02-06:
I checked my docker-compose.yml. There was not tabs. 
Here is my docker-compose for z2m:
# zigbee2mqtt
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - /home/admin/SmartHome/zigbee2mqtt/zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 7890 :7890
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_661b19c0d9c9eb11b2728b4f1d69213e-if00-port0:/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_661b19c0d9c9eb11b2728b4f1d69213e-if00-port0

Maybe i missed something.

loogfreak, 2023-02-06:
Now i can connect my devices und check the monitoring. Otherwise i´m not able to connect to my frontend. 
Thank you very much for your support!

dragondagda, 2023-02-07:
@loogfreak I see two issues with your port configuration:

1. don't use a space between 7890 and the colon
2. usually the frontend port is 8080. unless you made a manual configuration (see your configuration yaml -> is there a configuration for frontend.port?) you are forwarding the wrong port from the container to your host

so, I would suggest to set the ports section like this:
```
ports:
  - 7890:8080
```