Koenkk/zigbee2mqtt #16568: 1.30.1 only partially fix Tuya TS0601 _TZE200_lu01t0zl Human presence sensor with fall function

raphaabreu, 2023-02-07:
### What happened?

This is a continuation of the issue #16467.

The version 1.30.1 was released to fix it, however the fix has been only partial.

The "presence" field's type is boolean but the "occupancy" field's type is string:

[image]

I believe that this is the culprit:

https://github.com/Koenkk/zigbee-herdsman-converters/commit/348f9c001aa571950525570001373550d68f2a2a#diff-7ae23d3c13c2d7f7e37106590684c791aba8093f8be813b7c74a4f1484017c3bL3453-R3455

### What did you expect to happen?

I expect both sensors to be showing correctly on home assistant.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20210708

### Adapter

sonoff zigbee 3.0

Witek1974, 2023-02-07:
I confirm, this problem still exists, also in my setup.

Koenkk, 2023-02-07:
should be fixed in the latest dev. https://github.com/Koenkk/zigbee2mqtt/issues/16467

closed 2023-02-07 by Koenkk

christian1986, 2023-02-27:
I don't get any information about distance or precense. Is this the same issue then? Lux is updated.

"2023-02-27 15:14:59MQTT publish: topic 'zigbee2mqtt/Radar - Test', payload '{"cli":" ","detection_delay":0.1,"fading_time":75,"illuminance_lux":686,"last_seen":"2023-02-27T15:14:59+01:00","linkquality":94,"maximum_range":7.05,"minimum_range":0.15,"presence":false,"radar_sensitivity":8,"self_test":"check_success","target_distance":0}'"