Koenkk/zigbee2mqtt #16574: [New device support]: Add sensitivity support for LDSENK08

Chartreusito, 2023-02-07:
### Link

https://www.zigbee2mqtt.io/devices/LDSENK08.html#adeo-ldsenk08

### Database entry

{"id":42,"type":"EndDevice","ieeeAddr":"0x003c84fffe1b63d8","nwkAddr":48148,"manufId":4727,"manufName":"ADEO","powerSource":"Battery","modelId":"LDSENK08","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK08","manufacturerName":"ADEO","powerSource":3,"zclVersion":3,"appVersion":34,"stackVersion":6,"hwVersion":1,"dateCode":"20200519","swBuildId":"2.02"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0029e8a671","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8a671","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8a671","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":34,"stackVersion":6,"hwVersion":1,"dateCode":"20200519","swBuildId":"2.02","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1675448631711,"defaultSendRequestWhen":"active"}

### Comments

This door sensor is already supported but it's missing a functionnality.

According to device documentation, sensitivity can be set via the manufacturer gateway.
Process is similar as to what is documented here : https://www.zigbee2mqtt.io/devices/DJT11LM.html#sensitivity
Setting the sensitivity is supposed to be followed by an interaction to wake up the device.
[Documentation](https://media.adeo.com/marketplace/LMFR/82428867/3310687.pdf)

I tried with the following 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const fzLocal = {
    LDSENK08: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                contact: !((zoneStatus & 1) > 0),
                vibration: (zoneStatus & 1<<1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

const definition = {
    zigbeeModel: ['LDSENK08'],
    model: 'LDSENK08',
    vendor: 'ADEO',
    description: 'ENKI LEXMAN wireless smart door window sensor with vibration',
    fromZigbee: [fzLocal.LDSENK08, fz.battery],
    //fromZigbee: [fzLocal.LDSENK08, fz.battery, fz.sensitivity],
    toZigbee: [tz.TS0210_sensitivity],
    exposes: [e.battery_low(), e.contact(), e.vibration(), e.tamper(), e.battery(), exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])],
    //exposes: [e.battery_low(), e.contact(), e.vibration(), e.tamper(), e.battery(), e.sensitivity()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;
```

UsernetFR, 2023-02-21:
Same request. I'm getting false positives for vibration detection.
The sensor is compatible with zigbee2mqtt but it does not allow sensitivity adjustment for vibration.

zebu14, 2023-03-15:
Same request.
The default sensitivity seems a bit high.
Having a way to adjust sensitivity would be useful.

septeven, 2023-03-22:
Same request, particularly because it seems to help draining the battery faster.

Chartreusito, 2023-04-22:
This issue is still valid

Koenkk, 2023-04-23:
> According to device documentation, sensitivity can be set via the manufacturer gateway.

To add support for this, someone should sniff the traffic with the original gateway to figure out what command to send: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Chartreusito, 2023-05-24:
Issue still valid, i just got a CC2531 now i need to get the gateway for this device up and running

Chartreusito, 2023-06-24:
Still valid

Chartreusito, 2023-07-13:
I finally got the sniffer up and running, and i've got a dump of the moment i added the device and set sensitivity via the original Enki gateway, but i'm no expert with wireshark, any chance i can drop the dump somewhere for someone to have a look at ?

Koenkk, 2023-07-13:
@Chartreusito I expect that you will see a write request when setting the sensitivity, can you provide a screenshot of that here? Make sure to expand all the panels in the bottom which show the payload.

Koenkk, 2023-07-15:
- What sensitivity level did you set in the screenshot above, level 4 or 5?
- This ext converter should add support for the sensitivity: https://gist.github.com/Koenkk/f902e5e3f85c4e1624bac1133847c0fd

  - save this as file next to `configuration.yaml` as `ext_converter.js`
  - add it to `configuration.yaml`:
  ```yaml
  external_converters:
    - ext_converter.js
  ```
  - start z2m, check if issue is fixed

UsernetFR, 2023-07-15:
With this converter impossible to restart z2m :

```
[11:51:11] INFO: Preparing to start...
[11:51:11] INFO: Socat not enabled
[11:51:12] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

I already use another converter without problem

Koenkk, 2023-07-15:
Made a mistake, I updated the converter.

UsernetFR, 2023-07-15:
The converter already seems to work better.
A line appears with a new setting :

[image]

But when I indicate something (eg 2; 3; 4), I get a lot of error and the value does not save

[image]
[image]

Koenkk, 2023-07-15:
Did you wake-up the device right before changing it?

Chartreusito, 2023-07-16:
Just tried the external converter, values are saved when waking up device. The line has no name though, thanks !

Koenkk, 2023-07-17:
I updated https://gist.github.com/Koenkk/f902e5e3f85c4e1624bac1133847c0fd , can you check if it shows the description?

Chartreusito, 2023-07-17:
Description is now correctly showed, perfect, thank you.

Koenkk, 2023-07-18:
Integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-18 by Koenkk