Koenkk/zigbee2mqtt #16580: Sensor sensor.xxx has device class None, state class None and unit second thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>)

GerdRuetten, 2023-02-08:
### What happened?

With the latest stable Version i got several errors in the HA log:

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:608
Integration: Sensor (documentation, issues)
First occurred: 06:15:26 (4 occurrences)
Last logged: 06:15:26

Sensor sensor.handsender_pascal_action_duration has device class None, state class None and unit second thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Sensor sensor.zigbee_router_1_transmit_power has device class None, state class None and unit dBm thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Sensor sensor.handsender_marvin_action_duration has device class None, state class None and unit second thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Sensor sensor.zigbee_router_2_transmit_power has device class None, state class None and unit dBm thus indicating it has a numeric value; however, it has the non-numeric value: None (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

thebatfink, 2023-02-08:
I have this also. Its being reported by my Hue v1 Dimmer Switches https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621

GerdRuetten, 2023-02-08:
The four devices by me are:

2 x SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N), flashed as router
2 x [Philips 324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html)

Nardol, 2023-02-13:
It should be fixed with home-assistant/core#87004 so in Home Assistant 2023.3.

hellcry37, 2023-02-26:
same issue here, hope it get fixed in ha 2023.3

dpgh947, 2023-03-09:
Yes, I was just mentioning my tasmota issue as the problem here might be nothing to do with zigbee2mqtt either.

closed 2023-04-16 by github-actions[bot]

spanzetta, 2023-04-23:
I have the same issue with my sensor created from a Netatmo thermostat 

Sensor sensor.temp_casa has device class None, state class None and unit °C thus indicating it has a numeric value;

temp_casa:
        friendly_name: "Temperatura Casa"
        value_template: "{{states.climate.casa.attributes.current_temperature}}"
        unit_of_measurement: "°C"
        icon_template: mdi:thermometer

Spirituss, 2023-05-05:
The same is here

TonyM1958, 2023-05-14:
still a problem: Sensor sensor.battery_duration has device class None, state class None unit hours and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: Unavailable (<class 'str'>)

Runs fine once initialised, its just a template start up problem really but what else can you do to the template sensor apart from not giving it a device class to make this go away so I can run with clean logs?

emufan, 2023-05-20:
> really but what else can you do to the template sensor apart from not giving it a device class

Would be interested in this, too. A trigger template and check against a number is not working either. Where it should restore state on startup. But gives the same error.