Koenkk/zigbee2mqtt #16582: E1525 - Battery at 200%

weimdall, 2023-02-08:
### What happened?

The fix #16412 worked for the IKEA 5-buttons remote, but did not fix the displayed battery percentage of my E1525 motion sensor.
These two devices received the *same* update changing the battery value from 0-100 to 0-200.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20210708

### Adapter

zStack3x0 - Sonoff usb dongle

Koenkk, 2023-02-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-08 by Koenkk

weimdall, 2023-02-08:
Thank you,
I will check the dev build but I fear they will be a problem with E1745 motion sensors now :
Only the E1525 got its battery percentage multiply by two in my set up, E1745 are showing 0-100 range

Koenkk, 2023-02-09:
Isn't there an update available for them?

weimdall, 2023-02-09:
No, there is no update for E1745 (yet?)

The release note from IKEA ([https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html)) specify that the patch is only for Motion Sensors V1 (E1525)

weimdall, 2023-02-14:
@Koenkk I confirm that this is a bug in the dev version zigbee-herdsman-converters 15.0.46
The value of E1525 is fixed, but now the value of E1745 is divided by two

Koenkk, 2023-02-14:
Can you provide the data/database.db entry of the E1745?

weimdall, 2023-02-14:
Sure, here is the entry for my E1745

```json
{
  "id": 2,
  "type": "EndDevice",
  "ieeeAddr": "0xXXXXXXXXXXXXXXX",
  "nwkAddr": 65454,
  "manufId": 4476,
  "manufName": "IKEA of Sweden",
  "powerSource": "Battery",
  "modelId": "TRADFRI motion sensor",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 2128,
      "inClusterList": [
        0,
        1,
        3,
        9,
        32,
        4096,
        64636
      ],
      "outClusterList": [
        3,
        4,
        6,
        8,
        25,
        4096
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "TRADFRI motion sensor",
            "manufacturerName": "IKEA of Sweden",
            "powerSource": 3,
            "zclVersion": 3,
            "appVersion": 32,
            "stackVersion": 98,
            "hwVersion": 1,
            "dateCode": "20190308",
            "swBuildId": "2.0.022"
          }
        },
        "genPollCtrl": {
          "attributes": {
            "checkinInterval": 172800
          }
        },
        "genPowerCfg": {
          "attributes": {
            "batteryPercentageRemaining": 16
          }
        }
      },
      "binds": [
        {
          "cluster": 32,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0024c6c942",
          "endpointID": 242
        },
        {
          "cluster": 1,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0024c6c942",
          "endpointID": 242
        }
      ],
      "configuredReportings": [
        {
          "cluster": 1,
          "attrId": 33,
          "minRepIntval": 3600,
          "maxRepIntval": 62000,
          "repChange": 0
        }
      ],
      "meta": {}
    }
  },
  "appVersion": 32,
  "stackVersion": 98,
  "hwVersion": 1,
  "dateCode": "20190308",
  "swBuildId": "2.0.022",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 1
  },
  "lastSeen": 1676402095379,
  "defaultSendRequestWhen": "active"
}
```

Koenkk, 2023-02-15:
Pushed a possible fix, after updating to the latest-dev wait at least 24 hours to give the remote time to report its battery state.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

weimdall, 2023-02-17:
Seems to work!
I get expected battery level for E1745 et E1525