Koenkk/zigbee2mqtt #16583: Using native unit of measurement 'ppm' which is not a valid unit for the device class ('volatile_organic_compounds

cloudbr34k84, 2023-02-08:
### What happened?

I knwo this has been fixed for other devices, can this be fixed as well please

Entity sensor.laundry_climate_voc (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'ppm' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; expected one of ['µg/m³']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

### Zigbee2MQTT version

Zigbee2MQTT Edge

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator P

Koenkk, 2023-02-08:
@frenck `ppm` is a common unit, could this be supported?

frenck, 2023-02-08:
Nope, as we can't provide conversions for it.
We are considering a different device class for those, but for now, that isn't available yet.

Koenkk, 2023-02-09:
@frenck so maybe the best solution for now is to remove the `device_class` for these sensors?

cloudbr34k84, 2023-03-12:
Any solution to this?

closed 2023-04-19 by github-actions[bot]