Koenkk/zigbee2mqtt #16584: TV02 - OTA - No image available for imageType '5634'

seflm, 2023-02-08:
### What happened?

The OTA update for TV02-Zigbee device is not working with following error message.
Failed to check if update available for 'TRV - bedroom' (No image available for imageType '5634')

[image]

### How to reproduce it (minimal and precise)

Click on button to check OTA update for the TV02-Zigbee device.

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0

### Debug log

Zigbee2MQTT:info  2023-02-08 13:23:17: Checking if update available for 'TRV - bedroom'
Zigbee2MQTT:info  2023-02-08 13:23:17: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"TRV - bedroom"},"error":"Failed to check if update available for 'TRV - bedroom' (No image available for imageType '5634')","status":"error","transaction":"vd97q-3"}'
Zigbee2MQTT:error 2023-02-08 13:23:17: Failed to check if update available for 'TRV - bedroom' (No image available for imageType '5634')

Koenkk, 2023-02-08:
We don't have an OTA available for this device yet, to retrieve it follow https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html (requires TuYa bridge)

arsenyspb, 2023-03-11:
@Koenkk how about a wiki page with **Country | City | GitHub** gateway owners with Android studio already set-up listed, and maybe incentive (...coffee?..)? Those that buy various TuYa / SmartThings / Xiaomi ecosystem sensors outnumbers the owners of gateways, but they could then collaborate with willing community contributors?

closed 2023-04-19 by github-actions[bot]

lubomirp, 2023-11-05:
Looks like still no image available, is known any workaround how to update TRV with latest firmware without using of TuYa bridge?

Sn00kiT, 2023-11-08:
I ordered a cheap tuya Gateway and try to fetch the Firmware for you, If there is one available (-;

arsenyspb, 2023-11-08:
@Koenkk @Sn00kiT  hey I know you folks are busy 🙇 but maybe  really let's consider building a community Wiki page that I've suggested [in March](https://github.com/Koenkk/zigbee2mqtt/issues/16584#issuecomment-1464879942), so that owners of Tuya Gateways could be (with reasonable coffee chats) contacted...

lubomirp, 2023-11-08:
> I ordered a cheap tuya Gateway and try to fetch the Firmware for you, If there is one available (-;

@Sn00kiT which one you have ordered?

Koenkk, 2023-11-08:
@arsenyspb Im fine with this, could you create a PR for zigbee2mqtt.io?

Sn00kiT, 2023-11-08:
> > I ordered a cheap tuya Gateway and try to fetch the Firmware for you, If there is one available (-;
> 
> @Sn00kiT which one you have ordered?

I think this is irrelevant, as Long it contacts the tuya Servers and i can Access those later with the Android Emulator. I also have a Dev Account from tuya. Maybe i can fetch IT there somewhere. I ordered one from aubess with zigebee and Bluetooth and Type c. Maybe i can Hack this later with tasmota/esphome.

Sn00kiT, 2023-11-08:
> @arsenyspb Im fine with this, could you create a PR for zigbee2mqtt.io?

I think the zigbee2mqtt Database for devices is more than enough. Quick and dirty i have created a Telegramgroup for those in need of it
https://t.me/+Ec_kEj9GE7RjNjI0

Sn00kiT, 2023-11-16:
Just got my Gateway today, unfortnuately there is No Firmware Update available.[image]

Sn00kiT, 2023-11-16:
Adjusting the schedule to a very simple Plan Like 07:00/21 24:00/22 fixed my non heating Problem. I guess the Thermostat Just don't likes advanced Plans with more then 2 heating schedules.