Koenkk/zigbee2mqtt #16589: TS130F: No converter available for 'get' 'state' ()

menzbenz-jeedom, 2023-02-08:
### What happened?

In the zigbee2MQTT web interface many message error are diplayed coming from roller shutter switch. Currently I have 5 roller shutter switch and i receive every two minutes those message : 
Debug 2023-02-08 18:13:45Received MQTT message on 'zigbee2mqtt/Volet CH1/get' with data '{"state":""}'
Error 2023-02-08 18:13:45No converter available for 'get' 'state' ()
Debug 2023-02-08 18:13:45Received MQTT message on 'zigbee2mqtt/Volet CH2/get' with data '{"state":""}'
Error 2023-02-08 18:13:45No converter available for 'get' 'state' ()
Debug 2023-02-08 18:13:45Received MQTT message on 'zigbee2mqtt/Volet CH3/get' with data '{"state":""}'
Error 2023-02-08 18:13:45No converter available for 'get' 'state' ()
Debug 2023-02-08 18:13:45Received MQTT message on 'zigbee2mqtt/volet ch4/get' with data '{"state":""}'
Error 2023-02-08 18:13:45No converter available for 'get' 'state' ()
Debug 2023-02-08 18:13:45Received MQTT message on 'zigbee2mqtt/Volet Hall etage/get' with data '{"state":""}'
Error 2023-02-08 18:13:45No converter available for 'get' 'state' ()

### What did you expect to happen?

I exepect that I have not to get this message as the the shutter is not requested

### How to reproduce it (minimal and precise)

First, pair the shutter switch .
A few minutes later, error in the web console appears.
[image]

Here is the switch : 
[image]

The expose list:

[image]

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

6.10.0.0 build 169

### Adapter

Jeedom Atlas zigbee - EZSP v8

### Debug log

Debug 2023-02-08 18:23:43Received Zigbee message from 'porte garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":37888}' from endpoint 1 with groupID 0
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/Volet CH1/get' with data '{"state":""}'
Error 2023-02-08 18:23:46No converter available for 'get' 'state' ()
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/Volet CH2/get' with data '{"state":""}'
Error 2023-02-08 18:23:46No converter available for 'get' 'state' ()
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/Volet CH3/get' with data '{"state":""}'
Error 2023-02-08 18:23:46No converter available for 'get' 'state' ()
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/volet ch4/get' with data '{"state":""}'
Error 2023-02-08 18:23:46No converter available for 'get' 'state' ()
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/Volet Hall etage/get' with data '{"state":""}'
Error 2023-02-08 18:23:46No converter available for 'get' 'state' ()
Debug 2023-02-08 18:23:46Received MQTT message on 'zigbee2mqtt/Interrupteur Bubendorf/get' with data '{"state_l1":""}'
Debug 2023-02-08 18:23:46Publishing get 'get' 'state' to 'Interrupteur Bubendorf'
Debug 2023-02-08 18:23:46Received Zigbee message from 'Capteur de presence', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
Debug 2023-02-08 18:23:46Received Zigbee message from 'Capteur de presence', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
Info 2023-02-08 18:23:47MQTT publish: topic 'zigbee2mqtt/porte garage', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"porte garage","hardwareVersion":1,"ieeeAddr":"0xa4c13888f6b03730","manufacturerID":4417,"manufacturerName":"_TZE200_nklqjk62","model":"PJ-ZGD01","networkAddress":16830,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"garage_door_contact":true,"linkquality":120}'
Error 2023-02-08 18:24:06Publish 'get' 'state' to 'Interrupteur Bubendorf' failed: 'Error: Read 0xa4c138af057872da/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55977 - 1 - 40 - 6 - 1 after 10000ms)'
Debug 2023-02-08 18:24:06Error: Read 0xa4c138af057872da/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55977 - 1 - 40 - 6 - 1 after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Debug 2023-02-08 18:24:13Received Zigbee message from 'porte garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":38400}' from endpoint 1 with groupID 0
Info 2023-02-08 18:24:17MQTT publish: topic 'zigbee2mqtt/porte garage', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"porte garage","hardwareVersion":1,"ieeeAddr":"0xa4c13888f6b03730","manufacturerID":4417,"manufacturerName":"_TZE200_nklqjk62","model":"PJ-ZGD01","networkAddress":16830,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"garage_door_contact":true,"linkquality":128}'
Debug 2023-02-08 18:24:17Saving state to file /var/www/html/plugins/zigbee2mqtt/data/zigbee2mqtt/state.json
Debug 2023-02-08 18:24:43Received Zigbee message from 'porte garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":38912}' from endpoint 1 with groupID 0
Info 2023-02-08 18:24:47MQTT publish: topic 'zigbee2mqtt/porte garage', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"porte garage","hardwareVersion":1,"ieeeAddr":"0xa4c13888f6b03730","manufacturerID":4417,"manufacturerName":"_TZE200_nklqjk62","model":"PJ-ZGD01","networkAddress":16830,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"garage_door_contact":true,"linkquality":112}'
Debug 2023-02-08 18:25:03Received Zigbee message from 'porte garage', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
Debug 2023-02-08 18:25:13Received Zigbee message from 'porte garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
Info 2023-02-08 18:25:18MQTT publish: topic 'zigbee2mqtt/porte garage', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"porte garage","hardwareVersion":1,"ieeeAddr":"0xa4c13888f6b03730","manufacturerID":4417,"manufacturerName":"_TZE200_nklqjk62","model":"PJ-ZGD01","networkAddress":16830,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"garage_door_contact":true,"linkquality":144}'
Debug 2023-02-08 18:25:44Received Zigbee message from 'porte garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":39936}' from endpoint 1 with groupID 0
Info 2023-02-08 18:25:48MQTT publish: topic 'zigbee2mqtt/porte garage', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"porte garage","hardwareVersion":1,"ieeeAddr":"0xa4c13888f6b03730","manufacturerID":4417,"manufacturerName":"_TZE200_nklqjk62","model":"PJ-ZGD01","networkAddress":16830,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"garage_door_contact":true,"linkquality":132}'
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/Volet CH1/get' with data '{"state":""}'
Error 2023-02-08 18:25:49No converter available for 'get' 'state' ()
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/Volet CH2/get' with data '{"state":""}'
Error 2023-02-08 18:25:49No converter available for 'get' 'state' ()
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/Volet CH3/get' with data '{"state":""}'
Error 2023-02-08 18:25:49No converter available for 'get' 'state' ()
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/volet ch4/get' with data '{"state":""}'
Error 2023-02-08 18:25:49No converter available for 'get' 'state' ()
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/Volet Hall etage/get' with data '{"state":""}'
Error 2023-02-08 18:25:49No converter available for 'get' 'state' ()
Debug 2023-02-08 18:25:49Received MQTT message on 'zigbee2mqtt/Interrupteur Bubendorf/get' with data '{"state_l1":""}'
Debug 2023-02-08 18:25:49Publishing get 'get' 'state' to 'Interrupteur Bubendorf'

menzbenz-jeedom, 2023-03-06:
Hello experts, Have you any idea about this problem please? do you need more details ?
Thank you

closed 2023-04-14 by github-actions[bot]

piogundel, 2023-05-01:
Did you manage to overcome this problem? I got the same.

Sigri44, 2023-06-06:
Hello, got the same problem here !