Koenkk/zigbee2mqtt #16590: [Feature request] Starting zigbee2mqtt without MQTT server

schmurtzm, 2023-02-08:
### Is your feature request related to a problem? Please describe

Hi, I recently tested the last version of zigbee2mqtt (I had an experience with the first early versions) , it was awesome to see that there's now a web interface but the most awesome feature that I see is the possibility to create user extensions : with these javascript script, MQTT server is really optional : I just want to trigger an http API of my own but it still require a mqtt server which is totally useless in my case.
So I think that this issue should be opened again to serve the user extensions feature which are a real alternative to the MQTT server.
It could be useful to have access to the UI to configure all the stuffs (COM port), MQTT server... when it is not well configured.

### Describe the solution you'd like

Be able to run zigbee2mqtt when
- MQTT server is unavailable
- COM port is not well configured
- a user extension contains an error

### Describe alternatives you've considered

For now I install a useless MQTT server, fortunately the memory consumption of Mosquito is really small.

### Additional context

Tested on Windows
Zigbee2mqtt 1.301

danieledwardgeorgehitchcock, 2023-02-09:
The clue is sort of in the name of the project - Z2M uses MQTT as its backbone and it would take a lot of unpicking to change that.

The Zigbee Herdsman and Herdsman Converters projects give a closer representation to your needs - these are decoupled from Z2M and could be referenced and extended in any project.

You may want to look at the iobroker project which does just this.

junosuarez, 2023-03-12:
In the mean time, if you don't need to use the MQTT functionality but want a lower footprint than running Mosquitto, you can use a different MQTT server, like https://nanomq.io/ or https://www.npmjs.com/package/aedes

closed 2023-04-19 by github-actions[bot]

ghwaldo, 2024-07-20:
I agree with this request. Sometimes you are testing, and in my case while nr-server is updating, want to pair new devices.
And also, if the adpater is missing (i use zbbbridge pro) it should work further on and try to connect again.
Don't misunderstand, z2m is great work and i want to thank you for that.
But software, that is running like a deamon, should be error-tolerant as long as parts may work on. Here i.e. pairing of new devices works while mqtt-server is missing.