Koenkk/zigbee2mqtt #16595: [New device support]: Wireless switch with 4 buttons - _TZ3000_dziaict4 - TS0044

cloudbr34k84, 2023-02-09:
### Link

https://zigbee.blakadder.com/Tuya_4key.html

### Database entry

info  2023-02-09 15:29:20: MQTT publish: topic 'homeassistant/sensor/0xa4c13877de566574/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/4 Button Remote/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13877de566574"],"manufacturer":"TuYa","model":"Wireless switch with 4 buttons (TS0044)","name":"4 Button Remote"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"4 Button Remote action","state_topic":"zigbee2mqtt/4 Button Remote","unique_id":"0xa4c13877de566574_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'

### Comments

```
Friendly name
4 Button Remote
Description
Last seen
8 minutes ago
Availability
Online
Device type
EndDevice
Zigbee Model
TS0044
Zigbee Manufacturer
_TZ3000_dziaict4
Description
Wireless switch with 4 buttons
Support status
Supported

IEEE Address
0xa4c13877de566574
Network address
0x1393
Manufacturer
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
[TS0044](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044)
Power
Battery 
Interview completed
True

```
info  2023-02-09 15:29:20: MQTT publish: topic 'homeassistant/sensor/0xa4c13877de566574/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/4 Button Remote/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13877de566574"],"manufacturer":"TuYa","model":"Wireless switch with 4 buttons (TS0044)","name":"4 Button Remote"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"4 Button Remote action","state_topic":"zigbee2mqtt/4 Button Remote","unique_id":"0xa4c13877de566574_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
```

### External converter

```shell
No Converters
```

cloudbr34k84, 2023-02-10:
@Koenkk can wer get this added to the converter as it seems its already supported just not this model?

[image]

Koenkk, 2023-02-10:
```
Support status
Supported`
```

It is already detected as supported?

cloudbr34k84, 2023-02-10:
> ```
> Support status
> Supported`
> ```
> 
> It is already detected as supported?

I don't get any of the actions available in HA?

cloudbr34k84, 2023-02-10:
> ```
> Support status
> Supported`
> ```
> 
> It is already detected as supported?

[image]

[image]

Nothing is available

Koenkk, 2023-02-10:
Can you provide the debug log when pressing buttons on the device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cloudbr34k84, 2023-02-10:
> Can you provide the debug log when pressing buttons on the device?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.
[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/10711998/log.1.txt)

0xa4c13877de566574
TS0044
0x13A7

cloudbr34k84, 2023-02-13:
> > Can you provide the debug log when pressing buttons on the device?
> > See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.
> > [log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/10711998/log.1.txt)
> 
> 0xa4c13877de566574 TS0044 0x13A7

@Koenkk any updates mate?

Koenkk, 2023-02-14:
> > Can you provide the debug log when pressing buttons on the device?
> > See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.
> > [log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/10711998/log.1.txt)
> 
> 0xa4c13877de566574 TS0044 0x13A7

These are not the debug logs, see https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cloudbr34k84, 2023-02-14:
@Koenkk ahh ok, i have no idea when imn reading the instructions what or how to do that?? DO you have a better step by step guide for noobs? sorry

Koenkk, 2023-02-15:
You can find them in the `data/log` folder.

cloudbr34k84, 2023-02-16:
@Koenkk i dont see that. what am i missing here??
[image]

This is my config
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
  zigbee_herdsman_debug: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: XXXXXXXXXXXXX
  server: mqtt://core-mosquitto:1883
serial:
  port: /dev/ttyUSB0
availability: true
advanced:
  log_level: debug
  pan_id: GENERATE
  ext_pan_id:
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
  network_key: GENERATE
  channel: 25
homeassistant: true
discovery_topic: homeassistant
status_topic: hass/status
```

Koenkk, 2023-02-18:
The log files are in the `log` folder (just above `configuration.yaml`)

cloudbr34k84, 2023-02-19:
@Koenkk thats where i got the file from??
[image]

cloudbr34k84, 2023-02-19:
I have re added the device. I saw the below in the log
[log (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/10775806/log.2.txt)

info  2023-02-19 14:06:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wireless switch with 4 buttons","exposes":[{"access":1,"description":"Remaining battery in %, can take up to 24 hours before reported.","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["1_single","1_double","1_hold","2_single","2_double","2_hold","3_single","3_double","3_hold","4_single","4_double","4_hold"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0044","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"4 Button Remote","ieee_address":"0xa4c13877de566574","status":"successful","supported":true},"type":"device_interview"}'

Koenkk, 2023-02-19:
I don't see that in this log file (`log (2).txt`). I tried searching for `0xa4c13877de566574` but not results.

cloudbr34k84, 2023-02-19:
@Koenkk sorry try this pls
[0xa4c13877de566574_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10778255/0xa4c13877de566574_log.txt)

Koenkk, 2023-02-20:
I see that actions are being send:

```
debug 2023-02-20 09:20:52: Received Zigbee message from '0xa4c13877de566574', type 'raw', cluster 'genOnOff', data '{"data":[1,149,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
info  2023-02-20 09:20:52: MQTT publish: topic 'zigbee2mqtt/Bedroom 2 Climate', payload '{"co2":365,"formaldehyd":0,"humidity":55.4,"last_seen":"2023-02-19T23:20:52.212Z","linkquality":61,"pm25":8,"temperature":25.4,"voc":0}'
info  2023-02-20 09:20:52: MQTT publish: topic 'zigbee2mqtt/0xa4c13877de566574', payload '{"action":"1_single","battery":null,"last_seen":"2023-02-19T23:20:52.213Z","linkquality":102}'
info  2023-02-20 09:20:52: MQTT publish: topic 'zigbee2mqtt/0xa4c13877de566574', payload '{"action":"","battery":null,"last_seen":"2023-02-19T23:20:52.213Z","linkquality":102}'
info  2023-02-20 09:20:52: MQTT publish: topic 'zigbee2mqtt/0xa4c13877de566574/action', payload '1_single'
```

safonovi, 2023-02-20:
I have a same log. And no action on screen and HA
`
Debug 2023-02-20 22:32:43Received Zigbee message from '0xa4c1384dffc1cf60', type 'raw', cluster 'genOnOff', data '{"data":[1,20,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info 2023-02-20 22:32:43MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"4_single","battery":100,"last_seen":"2023-02-20T19:32:43.432Z","linkquality":120,"voltage":3000}'
Info 2023-02-20 22:32:43MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"","battery":100,"last_seen":"2023-02-20T19:32:43.432Z","linkquality":120,"voltage":3000}'
Info 2023-02-20 22:32:43MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60/action', payload '4_single'
Debug 2023-02-20 22:32:45Received Zigbee message from '0xa4c1384dffc1cf60', type 'raw', cluster 'genOnOff', data '{"data":[1,21,253,1],"type":"Buffer"}' from endpoint 4 with groupID 0
Info 2023-02-20 22:32:45MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"4_double","battery":100,"last_seen":"2023-02-20T19:32:45.726Z","linkquality":127,"voltage":3000}'
Info 2023-02-20 22:32:45MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"","battery":100,"last_seen":"2023-02-20T19:32:45.726Z","linkquality":127,"voltage":3000}'
Info 2023-02-20 22:32:45MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60/action', payload '4_double'
Debug 2023-02-20 22:32:49Received Zigbee message from '0xa4c1384dffc1cf60', type 'raw', cluster 'genOnOff', data '{"data":[1,22,253,1],"type":"Buffer"}' from endpoint 2 with groupID 0
Info 2023-02-20 22:32:49MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"2_double","battery":100,"last_seen":"2023-02-20T19:32:49.291Z","linkquality":123,"voltage":3000}'
Info 2023-02-20 22:32:49MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60', payload '{"action":"","battery":100,"last_seen":"2023-02-20T19:32:49.291Z","linkquality":123,"voltage":3000}'
Info 2023-02-20 22:32:49MQTT publish: topic 'zigbee2mqtt/0xa4c1384dffc1cf60/action', payload '2_double'
`
[image]

cloudbr34k84, 2023-02-20:
> I see that actions are being send:
>
> ```
> debug 2023-02-20 09:20:52: Received Zigbee message from '0xa4c13877de566574', type 'raw', cluster 'genOnOff', data '{"data":[1,149,253,0],"type":"Buffer"}' [...]

Yeah I saw that, however there is no 1_single in HA..

closed 2023-02-21 by cloudbr34k84

Koenkk, 2023-02-21:
The actions will always be empty in HA, see https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty