Koenkk/zigbee2mqtt #16597: Water irrigation valve TS0601 - _TZE200_a7sghmms - not responding to Switch State/Show On when its off

cloudbr34k84, 2023-02-09:
### What happened?

Having issues with using this device. I can sometimes turn it on with its switch state, and then i cant turn it off. Or i can turn if off but the state is not updated in Z2M. As you can see its on in Z2m and no water consumed has been picked up because its not actually on. I usually have to re add the device and it works again for while. 
[image]

### Zigbee2MQTT version

Zigbee2MQTT Edge

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator P

closed 2023-03-19 by github-actions[bot]

rfernalu, 2023-10-01:
Hi, i have the same problem. It's extremely unstable....

rfernalu, 2023-10-01:
Does anybody any idea or solution ?

cloudbr34k84, 2023-10-01:
> Does anybody any idea or solution ?
> 

What firmware are you using for that coordinator??

rfernalu, 2023-10-02:
Hi, the latest one. It's a SONOFF

cloudbr34k84, 2023-10-02:
> […](#)

whats the number you are running ie im below
[image]

rfernalu, 2023-10-05:
thanks!
check mine:
[image: image.png]

cloudbr34k84, 2023-10-05:
> thanks!
> check mine:
> [image: image.png]
> 
> 
> > Hi, the latest one. It's a SONOFF El dom, 1 oct 2023, 22:20, Brad *@*.
> >
> > *> escribió: … <#m_8342245210782672662_> Does anybody any idea or solution
> > ? What firmware are you using for that coordinator?? — Reply to this email
> > directly, view it on GitHub <#16597 (comment)
> > <https://github.com/Koenkk/zigbee2mqtt/issues/16597#issuecomment-1742190279>>,
> >
> > whats the number you are running ie im below
> > [image: image]
> > <https://user-images.githubusercontent.com/58960644/271912933-5850e294-ec15-4c43-939b-81438c51a943.png>
> >
> >
> 
> 
> Raul.-

No image?