Koenkk/zigbee2mqtt #16605: TypeError: The "path" argument must be of type string. Received undefined -> at writeDevicesOrGroups (/app/lib/util/settings.ts:245:38)

zeroflag0, 2023-02-09:
### What happened?

When starting zigbee2mqtt (or sometimes it will start running but then bring the error when e.g. pairing a device) it will throw the error and it will either stop completely or continue in a non-working state (then mqtt works, zigbee doesn't)

```
[00:05:12] INFO: [32mZigbee Herdsman debug logging enabled[0m
[00:05:14] INFO: [32mStarting Zigbee2MQTT...[0m
Zigbee2MQTT:debug 2023-02-09 00:05:18: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-02-09 00:05:18: Logging to console and directory: '/config/zigbee2mqtt/log/2023-02-09.00-05-18' filename: log.txt
Zigbee2MQTT:info  2023-02-09 00:05:18: Starting Zigbee2MQTT version 1.30.1 (commit #unknown)
Zigbee2MQTT:info  2023-02-09 00:05:18: Starting zigbee-herdsman (0.14.89)
Zigbee2MQTT:debug 2023-02-09 00:05:18: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2659381-if00"}}'
2023-02-08T23:05:19.280Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-02-08T23:05:19.284Z zigbee-herdsman:deconz:driver Set delay to 0
2023-02-08T23:05:19.285Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2659381-if00","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-02-08T23:05:19.286Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2659381-if00
2023-02-08T23:05:19.302Z zigbee-herdsman:deconz:driver Serialport opened
2023-02-08T23:05:19.303Z zigbee-herdsman:controller:log Started with result 'resumed'
2023-02-08T23:05:19.303Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2023-02-08T23:05:19.350Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2023-02-08T23:05:19.362Z zigbee-herdsman:deconz:frameParser MAC: 0x00212effff0956a9
2023-02-08T23:05:19.365Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7
2023-02-08T23:05:19.368Z zigbee-herdsman:deconz:frameParser NWKADDR: 0
Zigbee2MQTT:info  2023-02-09 00:05:19: zigbee-herdsman started (resumed)
2023-02-08T23:05:19.380Z zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3
2023-02-08T23:05:19.383Z zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,114,38
Zigbee2MQTT:info  2023-02-09 00:05:19: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
2023-02-08T23:05:19.391Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5
2023-02-08T23:05:19.393Z zigbee-herdsman:deconz:frameParser PANID: 1a62
2023-02-08T23:05:19.396Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11
2023-02-08T23:05:19.399Z zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd
2023-02-08T23:05:19.402Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28
2023-02-08T23:05:19.405Z zigbee-herdsman:deconz:frameParser CHANNEL: 11
2023-02-08T23:05:19.408Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24
2023-02-08T23:05:19.410Z zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden
Zigbee2MQTT:debug 2023-02-09 00:05:19: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:error 2023-02-09 00:05:19: Failed to start zigbee
Zigbee2MQTT:error 2023-02-09 00:05:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-09 00:05:19: Exiting...
Zigbee2MQTT:error 2023-02-09 00:05:19: TypeError: The "path" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:387:5)
    at validateString (node:internal/validators:121:11)
    at Object.resolve (node:path:1098:7)
    at Object.joinPath (/app/lib/util/data.ts:17:17)
    at writeDevicesOrGroups (/app/lib/util/settings.ts:245:38)
    at write (/app/lib/util/settings.ts:250:5)
    at Object.addDevice (/app/lib/util/settings.ts:555:5)
    at Device.ensureInSettings (/app/lib/model/device.ts:41:22)
    at Zigbee.resolveDevice (/app/lib/zigbee.ts:234:20)
    at /app/lib/zigbee.ts:276:30

```

### What did you expect to happen?

no exception

### How to reproduce it (minimal and precise)

I've had this the first time playing around with hass (hass-os on raspi4) and zigbee when switching from ZHA to z2m. this time I'm switching from ZHA -> deCONZ -> z2m.

Suspecting that the usb-device is not correctly connected I've tried ["fails to start" to verify the port](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port) but it reports success. I can't say if that's the case from inside the docker because it usually stops before I can bash into it. (no idea how to do that with docker and hass)

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

202205...

### Adapter

Conbee II

zeroflag0, 2023-02-11:
After 3 hours of debugging I found the issue, but cannot fix it because I'm too newbish with ts...

TL;DR: workaround -> completely remove empty `devices: []` and `groups: []` from `configuration.yaml`

```yaml
devices: []
```
in classic yaml that would be valid.
however z2m supports 2 variants of devices/groups.
```yaml
devices: 
  '0x1234567890':
    friendly_name: foo
  '0x123456789f':
    friendly_name: bar
```
this is an array and read and handled as such.

however the second option is a string:
```yaml
devices: filename.yaml
```
which would lead to a second file which then contains the array.

now comes the unclean part: the type-check is ambiguous, empty-array-tostring gives us "" and we end up with a filename that cannot be opened.
[lib/util/settings.ts]
```ts
const writeDevicesOrGroups = (type: 'devices' | 'groups'): void => {
    if (typeof actual[type] === 'string' || Array.isArray(actual[type])) {
        const fileToWrite = Array.isArray(actual[type]) ? actual[type][0] : actual[type];
```
(btw why doesn't anyone put that poor `actual[type]` into a variable?)

I think somewhere there with the type-checks is the issue, but my ts-knowlege is like 3 hours deep, just a thought.

Could someone with real ts knowledge find the exact way to not load [] as a filename?

zeroflag0, 2023-02-11:
2 more hours into ts and I think I found a way to fix it in code. (even though I haven't found the actual root cause yet)
https://github.com/Koenkk/zigbee2mqtt/pull/16629

It seems during read() the code needs NOT to go through readDevicesOrGroups with an empty array to succeed.
I've done that by checking the array for it's length. (it's actually an interface but Array.isArray is true?)

Koenkk, 2023-02-13:
Adressed in https://github.com/Koenkk/zigbee2mqtt/pull/16629

closed 2023-02-13 by Koenkk

Maxcrouz73, 2023-02-23:
Hello,

Same situation here, i have try modify the code following your commit but with no success...
I just cannot add a group or rename a device, i run into the same error:

zigbee2MQTT:error 2023-02-23 12:05:59: Request zigbee2mqtt/bridge/request/device/rename failed with error: The "path" argument must be of type string. Received undefined

Just don't know what to do, because before yesterday everythings was working just fine since almost a year.

bertheto, 2023-02-25:
hello same issue on my side ...

bertheto, 2023-02-25:
hey, seemingly it's fixed in the 1.30.1-dev version ,thaks to the devs !