Koenkk/zigbee2mqtt #16608: The power, energy, device_temperature, voltage, and current of Aqara smart wall outlet T1 are N/A, and the setting of led_disabled_night is invalid.

maboroshina, 2023-02-09:
### What happened?

I paired new devices but these are empty
Try looking at the debug log, it doesn't seem to actively send power, energy, etc.
[image]

But I can read uncalibrated power in Dev console
like this
[image]

Here are the device details
[image]

### What did you expect to happen?

display value

### Zigbee2MQTT version

 1.30.0 commit: 8f781db

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

<details><summary>The log is too long, so fold it</summary>
<p>

```
Info 2023-02-10 04:17:24Device '0x04cf8cdf3c7f4a4c' joined
Info 2023-02-10 04:17:24MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7f4a4c","ieee_address":"0x04cf8cdf3c7f4a4c"},"type":"device_joined"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c7f4a4c/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c/set","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"name":"0x04cf8cdf3c7f4a4c","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","name":"0x04cf8cdf3c7f4a4c power","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"device_class":"energy","enabled_by_default":true,"name":"0x04cf8cdf3c7f4a4c energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/device_temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","name":"0x04cf8cdf3c7f4a4c device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0x04cf8cdf3c7f4a4c voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","name":"0x04cf8cdf3c7f4a4c current","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c7f4a4c/switch_power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"entity_category":"config","icon":"mdi:memory","name":"0x04cf8cdf3c7f4a4c power_outage_memory","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_switch_power_outage_memory_zigbee2mqtt","value_template":"{% if value_json.power_outage_memory %} true {% else %} false {% endif %}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c7f4a4c/switch_led_disabled_night/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c/set/led_disabled_night","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"entity_category":"config","icon":"mdi:led-off","name":"0x04cf8cdf3c7f4a4c led_disabled_night","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_switch_led_disabled_night_zigbee2mqtt","value_template":"{% if value_json.led_disabled_night %} true {% else %} false {% endif %}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c7f4a4c/switch_button_lock/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c/set/button_lock","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"entity_category":"config","icon":"mdi:lock","name":"0x04cf8cdf3c7f4a4c button_lock","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_switch_button_lock_zigbee2mqtt","value_template":"{{ value_json.button_lock }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7f4a4c/overload_protection/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"enabled_by_default":false,"name":"0x04cf8cdf3c7f4a4c overload protection","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_overload_protection_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.overload_protection }}"}'
Info 2023-02-10 04:17:24MQTT publish: topic 'homeassistant/number/0x04cf8cdf3c7f4a4c/overload_protection/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c/set/overload_protection","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7f4a4c"],"manufacturer":"Xiaomi","model":"Aqara smart wall outlet T1 (QBCZ14LM)","name":"0x04cf8cdf3c7f4a4c"},"max":2500,"min":100,"name":"0x04cf8cdf3c7f4a4c overload protection","state_topic":"zigbee2mqtt/0x04cf8cdf3c7f4a4c","unique_id":"0x04cf8cdf3c7f4a4c_overload_protection_zigbee2mqtt","unit_ [... cut ...]
Info 2023-02-10 04:19:12MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7f4a4c', payload '{"button_lock":null,"current":null,"device_temperature":null,"energy":null,"led_disabled_night":null,"linkquality":160,"overload_protection":null,"power":null,"power_outage_memory":null,"state":"ON","update":{"installed_version":50,"latest_version":50,"state":"idle"},"update_available":null,"voltage":null}'
Debug 2023-02-10 04:19:12Responded to OTA request of '0x04cf8cdf3c7f4a4c' with 'NO_IMAGE_AVAILABLE'
```

</p>
</details>

MrBamse, 2023-02-20:
Same here, but with the Smart Switch.
Zigbee2MQTT: 1.30.1 commit: eb878d3
Coordinator type: zStack3x0
Coordinator revision: 20210708
Coordinator: Sonoff Universal Zigbee 3.0 USB Dongle Plus 
Device: https://www.zigbee2mqtt.io/devices/WS-EUK03.html#xiaomi-ws-euk03
(Im running the device with neutral)

Attributes with null: action, power, energy, device_temperature, power_outage_count
What I've done so far: Removed it in Z2M, restarted Z2M, reset the switch (long press 5s) but unfortunately with same result.

If I click the blue button for "refresh" of the attribute power, I get the following error message:
Error 2023-02-20 21:25:12Publish 'get' 'power' to 'Kitchen Ceiling' failed: 'TypeError: Cannot read properties of undefined (reading 'read')'

[image]

Koenkk, 2023-02-21:
The firmware you are using on the coordinator is too old, update to the latest one and re-pair the device.

MrBamse, 2023-02-21:
I can confirm that upgrading the FW solved the issue (had to remove, restart and readd the device post fw upgrade ofc).

For anyone it might help, this is how i upgraded the FW on my SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P:
Download and unpack the firmware (.hex) from: https://www.zigbee2mqtt.io/guide/adapters/#recommended

Shutdown Z2M and HA
git clone https://github.com/JelmerT/cc2538-bsl
cd cc2538-bsl
python3 setup.py
python3 cc2538-bsl.py -ewv --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Done, start/reboot everything

closed 2023-02-22 by Koenkk

ray17n, 2023-02-23:
Hi, I am having the same issue reported by @MrBamse. 

I updated the coordinator to the last revision (20221226) following the same steps shared by @MrBamse (thanks!), but the power and energy values are still reported as "Null". 

The Zigbee2MQTT version is the 1.30.1.

I restarted Home Assistant and Zigbee2MQTT a couple of times but it doesn't affect it.

Is there any additional steps that should be done?

MrBamse, 2023-02-24:
@ray17n sorry if this has been done but I really just want to make sure since it is super crucial. Did you delete/force delete the device in Z2M and then restart Z2M before trying to readd it?

ray17n, 2023-02-24:
@MrBamse I see, I didn't do it yet, I only rebooted it (I was afraid of losing all the existing pairings, I have a quite populated network). I was thinking to do it next week that I have more time to setup everything again if it's needed. I'll update the post with my results. Thanks.

maboroshina, 2023-03-09:
I wrote a custom converter to read uncalibrated data, and when I thought that was the only way to do it, since one day it started reporting full data. This is weird, I didn't upgrade the firmware, or do anything else, and one day I found out that my converter wasn't working, checked the logs, and found it was reporting full data normally, so deleted the custom converter, and it worked fine ,what happened?