Koenkk/zigbee2mqtt #16609: Last Seen setting enabled breaks Inovelli VZM31-SN get exposes from device

mbrink123, 2023-02-10:
### What happened?

After installing a new Inovelli Blue switch (vzm31-sn) all the settings on the exposes tab of the device in Z2M say null.  There is a blue refresh button that pulls the config from the switch and updates Z2M.  If Z2M has the Settings--Advanced--Last Seen setting set to ISO_8601_local the blue refresh button wont pull any data from the switch.  If you disable the last seen setting, you then get data when clicking the blue refresh button.  Enabling the last seen setting will again break pulling data from the switch. 

### What did you expect to happen?

When clicking the blue refresh button, it should pull whatever data is in the Inovelli switch and display the information in Z2M on the exposes tab.

### How to reproduce it (minimal and precise)

Change the "last seen" setting in Z2M (settings--advanced--last seen) to ISO_8601_local, add a new inovelli switch to Z2M, go to the device page for that switch, goto the exposes tab, scroll down to any setting that is blank and click the refresh button (nothing will happen).  Go back into settings and set the "last seen" setting to "disable", go back to the device exposes tab and click the refresh, the data will adjust to what's in the switch. 

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

 SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

ZetaWaves, 2023-03-06:
I am experiencing the same problem. It causes Z2M to constantly send payloads requesting the information that isn't specifically set, even if it's set as default values on the switch itself. This needs fixed ASAP.

ZetaWaves, 2023-04-06:
Bump. Still a problem.

mbrink123, 2023-05-08:
It's still needed.

mbrink123, 2023-06-09:
It is still an issue.

bradsjm, 2023-06-17:
Bump, same issue.

mbrink123, 2023-07-18:
It still needs fixed

mbrink123, 2023-08-19:
Unstale

Koenkk, 2023-08-19:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

mbrink123, 2023-08-22:
I think this is what your looking for.

mbrink123, 2023-09-22:
Unstale

Koenkk, 2023-09-23:
I don't see any log file in https://github.com/Koenkk/zigbee2mqtt/issues/16609#issuecomment-1688705489

mbrink123, 2023-09-27:
I guess email replies dont attach attachments.  Lets try again.
[45df7312_zigbee2mqtt_2023-08-22T18-26-06.181Z(1).log](https://github.com/Koenkk/zigbee2mqtt/files/12741339/45df7312_zigbee2mqtt_2023-08-22T18-26-06.181Z.1.log)

Koenkk, 2023-09-27:
I see the log level is not set to debug, also can you make sure you reproduce the issue in the log?

> Change the "last seen" setting in Z2M (settings--advanced--last seen) to ISO_8601_local, add a new inovelli switch to Z2M, go to the device page for that switch, goto the exposes tab, scroll down to [...]

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

mbrink123, 2023-09-27:
this work?  Look at office light switch at the very end of the log.
[45df7312_zigbee2mqtt_2023-09-27T18-27-18.462Z.log](https://github.com/Koenkk/zigbee2mqtt/files/12742057/45df7312_zigbee2mqtt_2023-09-27T18-27-18.462Z.log)

Koenkk, 2023-09-27:
I don't see you doing this in the log:

> scroll down to any setting that is blank and click the refresh button (nothing will happen)

raineth, 2023-10-05:
Here are 2 requests:
```
debug 2023-10-05 04:28:27: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"ledIntensityWhenOff":""}'
debug 2023-10-05 04:28:27: Publishing get 'get' 'ledIntensityWhenOff' to 'Master Bedroom Light'
debug 2023-10-05 04:28:27: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"ledIntensityWhenOff":1}' from endpoint 1 with groupID 0
info  2023-10-05 04:28:27: MQTT publish: topic 'zigbee2mqtt/Master Bedroom Light', payload '{[...TRIMMED...],"last_seen":"2023-10-05T04:28:27-04:00",[...TRIMMED...],"ledIntensityWhenOff":null,[...TRIMMED...]}'
debug 2023-10-05 04:28:42: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"ledIntensityWhenOff":""}'
debug 2023-10-05 04:28:42: Publishing get 'get' 'ledIntensityWhenOff' to 'Master Bedroom Light'
debug 2023-10-05 04:28:42: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"ledIntensityWhenOff":1}' from endpoint 1 with groupID 0
info  2023-10-05 04:28:42: MQTT publish: topic 'zigbee2mqtt/Master Bedroom Light', payload '{[...TRIMMED...],"last_seen":"2023-10-05T04:28:42-04:00",[...TRIMMED...],"ledIntensityWhenOff":null,[...TRIMMED...]}'
```

[Untrimmed log messages attached.](https://github.com/Koenkk/zigbee2mqtt/files/12814318/getrequest.txt)

Koenkk, 2023-10-05:
@raineth I see that the state is published correctly. Can you provide a debug log of the following:

- Disable last seen
- Do a `zigbee2mqtt/Master Bedroom Light/get`
- -> Should work (`MQTT publish...`)
- Enable last seen
- Do a `zigbee2mqtt/Master Bedroom Light/get`
- -> Should **not** work (`MQTT publish...`)

raineth, 2023-10-05:
I'm seeing the inverse of your expectation.  For clarity, I removed the cached state for the attribute I'm testing with (ledIntensityWhenOff) and restarted zigbee2mqtt before doing this test.

```
debug 2023-10-05 15:37:27: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{[...],"last_seen":"disable",[...]}},"transaction":"h2hzx-1"}'
info  2023-10-05 15:37:27: Successfully changed options
info  2023-10-05 15:37:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"h2hzx-1"}'
debug 2023-10-05 15:37:38: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"ledIntensityWhenOff":""}'
debug 2023-10-05 15:37:38: Publishing get 'get' 'ledIntensityWhenOff' to 'Master Bedroom Light'
debug 2023-10-05 15:37:39: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"ledIntensityWhenOff":1}' from endpoint 1 with groupID 0
debug 2023-10-05 15:38:10: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{[...],"last_seen":"ISO_8601_local",[...]}},"transaction":"h2hzx-2"}'
info  2023-10-05 15:38:10: Successfully changed options
info  2023-10-05 15:38:10: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"h2hzx-2"}'
debug 2023-10-05 15:38:14: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"ledIntensityWhenOff":""}'
debug 2023-10-05 15:38:14: Publishing get 'get' 'ledIntensityWhenOff' to 'Master Bedroom Light'
debug 2023-10-05 15:38:14: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"ledIntensityWhenOff":1}' from endpoint 1 with groupID 0
info  2023-10-05 15:38:14: MQTT publish: topic 'zigbee2mqtt/Master Bedroom Light', payload '{[...]"last_seen":"2023-10-05T15:38:14-04:00",[...],"ledIntensityWhenOff":1,[...]}'
```

[Full log](https://github.com/Koenkk/zigbee2mqtt/files/12822723/bug11609-2.txt)

Koenkk, 2023-10-07:
@raineth can you provide the contents of you data/configuration.yaml (without passwords/ips in it)

raineth, 2023-10-09:
Sure, here's my [redacted configuration.yaml](https://github.com/Koenkk/zigbee2mqtt/files/12842347/configuration.yaml.txt)

Koenkk, 2023-10-09:
@raineth pushed a fix

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

raineth, 2023-10-11:
I'm at HEAD for zigbee2mqtt.  I tried updating my package.json to bump the version of zigbee-herdsman-converters to 15.91.0, ran npm install/ci, but zigbee2mqtt doesn't start:
```
> zigbee2mqtt@1.33.1 start
> node index.js

Error: Cannot find module '/srv/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:409:19)
    at Function.Module._findPath (node:internal/modules/cjs/loader:651:18)
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1025:27)
    at Function.Module._load (node:internal/modules/cjs/loader:885:27)
    at Module.require (node:internal/modules/cjs/loader:1105:19)
    at require (node:internal/modules/cjs/helpers:103:18)
    at Object.<anonymous> (/srv/zigbee2mqtt/lib/model/device.ts:3:1)
    at Module._compile (node:internal/modules/cjs/loader:1218:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1272:10)
    at Module.load (node:internal/modules/cjs/loader:1081:32)
```
It's possible I've missed a step, I haven't done any serious NodeJS development.

Koenkk, 2023-10-11:
Don't do any manual changes, instead execute `npm install zigbee-herdsman-converters@15.92.1`

raineth, 2023-10-11:
Cool, that worked!  Refreshing integer settings now works properly and update in the UI as soon as the Zigbee response is received.

I tried refreshing other values, and I discovered that changing non-integer settings seems to be broken now regardless of the last_seen setting.

Some examples:
- Refreshing InvertSwitch caused my cached value to be (apparently) cleared
- Refreshing PowerType caused the value to be replaced with "1" instead of "Neutral"
- Refreshing LoadLevelIndicatorTimeout caused the selected value to be replaced with "----"

I tried downgrading zigbee-herdsman-converters back to 15.86.1 and was able to refresh those values if last_seen was disabled, although they didn't update in the UI until I reloaded the page.

15.92.1
```
debug 2023-10-11 16:34:36: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"powerType":""}'
debug 2023-10-11 16:34:36: Publishing get 'get' 'powerType' to 'Master Bedroom Light'
debug 2023-10-11 16:34:36: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"powerType":1}' from endpoint 1 with groupID 0
info  2023-10-11 16:34:36: MQTT publish: topic 'zigbee2mqtt/Master Bedroom Light', payload '{...,"powerType":1,...}'
```

vs.

15.86.1
```
debug 2023-10-11 16:15:16: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light/get' with data '{"powerType":""}'
debug 2023-10-11 16:15:16: Publishing get 'get' 'powerType' to 'Master Bedroom Light'
debug 2023-10-11 16:15:16: Received Zigbee message from 'Master Bedroom Light', type 'readResponse', cluster 'manuSpecificInovelliVZM31SN', data '{"powerType":1}' from endpoint 1 with groupID 0
info  2023-10-11 16:18:05: MQTT publish: topic 'zigbee2mqtt/Master Bedroom Light', payload '{...,"powerType":"Neutral",...}'
```

Koenkk, 2023-10-12:
Should be fixed now

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

raineth, 2023-10-12:
Everything seems to be working properly with zigbee-herdsman-converters@15.93.0, thank you!

closed 2024-05-10 by github-actions[bot]